florianfesti/boxes #159: svg to Coral Draw

Davejackaman, 2019-08-03:
Hi
I have been using the boxes program very successfully for the past two year.
Suddenly I have the following problem. I design a box and then save it into a file, When I try to open the file with Coral 17 it opens the file but the screen is blank. This was working well up to two weeks ago.
I now open it in Inkscape, copy the file and paste it into Coral, the result is not very good.
My PC is Win 10 and I am using Goral 17 and Coral Laser.
I have no experience or idea about coding or changing programs. Please help
avocad@telkomsa.net regards
Dave

Blancanegra, 2019-08-05:
Have you tried different formats?... AI, PDF and PS should work in Corel Draw.

florianfesti, 2019-08-05:
Can you try this SVG please (after ungzipping). Sorry for the inconvenience but GH does not allow uploading plain SVGs.
[ClosedBox.svg.gz](https://github.com/florianfesti/boxes/files/3467131/ClosedBox.svg.gz)

Looks like the new SVG post processing code has still issues. I am actually surprised the SVGs work in the browser and Inkscape. Some new (not yet rolled out) patch 73422b0faf0665ee4145499bc6b86af6921fb876 fix the XML name spaces with got messed up.

Blancanegra, 2019-08-05:
> Can you try this SVG please (after ungzipping). Sorry for the inconvenience but GH does not allow uploading plain SVGs.
> [ClosedBox.svg.gz](https://github.com/florianfesti/boxes/files/3467131/ClosedBox.svg.gz)
> 
> Looks like the new SVG post processing code has still issues. I am actually surprised the SVGs work in the browser and Inkscape. Some new (not yet rolled out) patch [73422b0](https://github.com/florianfesti/boxes/commit/73422b0faf0665ee4145499bc6b86af6921fb876) fix the XML name spaces with got messed up.

I tried it in Corel Draw 14 and it works, it's a 10x10x10 cm closed box.

florianfesti, 2019-08-05:
OK, I updated the web instance. Please re-open if the issue persists.

closed 2019-08-05 by florianfesti

Davejackaman, 2019-08-06:
Hi. Thank you for all the advice and help. I was away guiding a safari and got back this morning. Now I can not get into www.festi.info/boxes.py/ it says "This site can't be reached"  Sorry but I am not very computer literate and appreciate your help. Regards. Dave