florianfesti/boxes #164: Webpage problem

Tagej, 2019-08-16:
It looks linke the webpage Boxes.py have a problem, I can’t génerate dxf files. When downloaded they don’t have any content (0 KB). AI files become PS files, and I get a lot og servererrors when generate files
Tage

florianfesti, 2019-08-16:
Yeah, looks like pstoedit broke again. Or more precisely ghostscript broke it.
See https://sourceforge.net/p/pstoedit/bugs/25/?limit=25 for upstream details.

I filed https://bugzilla.redhat.com/show_bug.cgi?id=1741716 in the hope someone updates the package to fix this issue.
I may downgrade the web instance in the mean time *sigh*

florianfesti, 2019-08-16:
Looks like going back to ghostscript-9.25 solves the issue.
Downgrade the we instance. Please test if everything is back to normal.

Leaving this ticket open as a reminder to go back to the newest version as soon as this issue is resolved on the distribution level.

Tagej, 2019-08-17:
Hi Florian

You work fast - It looks like everything is back to normal.
but maybe there's a bug about fileextensions, when I generates a DXF file, the file have the right content but no fileextension. ie it generate "AngledBox" and I have to rename it to "AngledBox.dxf". There's the same issue with PLT files. 
(I use win10)

florianfesti, 2019-08-17:
Turns out adding one http header does the trick. Guess it is easy to tell I am not really a web developer...

Tagej, 2019-08-18:
It's not quite there yet 
I'm getting: "ERR_RESPONSE_HEADERS_MULTIPLE_CONTENT_DISPOSITION" when trying to generate a DXF or PLT
(I'm using Chrome Browser)

florianfesti, 2019-08-18:
OK, this should be fixed now. I added the headers to some global values instead of manipulating a copy. Turns out you really only want one file name to download to...

Tagej, 2019-08-18:
Brilliant - everything seems to work
Thank you very much - I'm going to lasercut some boxes with my students next week

With best regards
Tage

florianfesti, 2021-05-22:
Ok, at least in Fedora there are newer versions now. Let's assume other distros are at a similar state. Closing.

closed 2021-05-22 by florianfesti