florianfesti/boxes #165: Strange RegularBox corners

Durss, 2019-08-17:
Hello !

First of all, thank you for that amazing work !

I just have one question about the RegularBox template ( https://www.festi.info/boxes.py/RegularBox )

Is there a specific reason why notches' corners are made out of 3 lines, one of them being curved ?
[image]

Regards,
François

gcollic, 2019-08-18:
Hello François,
(I'm just a user like you, but I can answer your question)
The full explanation is here: https://florianfesti.github.io/boxes/html/api_burn.html
The short version is that the laser beam is a small circle, but a circle nonetheless. If the angle generated was a perfect angle, the angle cut would be round, and might hinder assembly. Boxes.py knows if the angle are "inside" or "outside", and prefers to cut without "undercut". This adds a small overcut to compensate the round nature of the laser beam. It happens automatically for all the boxes, and is managed by the `burn` setting (aka the kerf).
Regards,
Guillaume

Durss, 2019-08-18:
Ooh awesome !
Thank you for clarifying this :)

I close the thread.

closed 2019-08-18 by Durss