florianfesti/boxes #170: Inkscape extension errors in windows

jjisnow, 2019-09-08:
Several hours of working out inkscape 0.92.4 and trying to make it work with boxes, I've figured out the following.

1. inkscape uses some version of python 2.x which still uses iteritems() to return data from dictionaries. I know because I over-rode the setting in preferences.xml "python-interpreter" which then broke ALL the extensions in the menu, trying to get boxes to work.

2. there is now the error associated with 
"from pkg_resources import get_distribution" in boxes.py

Traceback (most recent call last):
  File "boxes.py", line 21, in <module>
    from pkg_resources import get_distribution
ImportError: No module named pkg_resources

any ideas on this?

jjisnow, 2019-09-08:
ok, so commenting out line 21 works, but then it throws the error 

`  File "boxes/__init__.py", line 23, in <module>
    import cairo
ImportError: No module named cairo`

I suspect the whole inkscape, internally locked python 2.7, not happy with python thing is limiting it's application to inkscape

jjisnow, 2019-09-08:
Duplicate of #47, I think

florianfesti, 2019-09-08:
Sorry, Boxes.py and Windows is still a bit of a difficult topic. #47 has a few pointers.
As far as I understand Inkscape should work with Boxes.py running on an Python 3.7 installation as it does just execute the `boxes` script. But with Windows lacking `#!` support that's also kinda icky.

florianfesti, 2021-05-22:
Closing this ticket. Things have moved on. See #314 for the current discussion on building and Inkspace.

closed 2021-05-22 by florianfesti