florianfesti/boxes #172: Unexpected behaviour of TrayLayout on command line

PellelNitram, 2019-09-21:
Hi,

I try to write my first own generator and started off by using TrayLayout as a base.

During that process, I found that:

1. ``boxes TrayLayout`` returns an error. The error is ``AttributeError: 'TrayLayout' object has no attribute 'format'`` and might be connected to the fact that I didn't specify any CL arguments. However, from its code I would expect it to work without CL arguments as default values are properly set.

2. Then I tried to set a value for ``--x`` with ``boxes TrayLayout --x 3``. Also for that, I got unexpected behaviour in the form of the following return: ``boxes: error: argument --x: expected one argument``. I would expect that the ``3`` is recognised as one argument for ``x``.

Have you encountered similar issues? Or am I simply silly by using the code incorrectly? Thanks!

florianfesti, 2019-09-21:
TrayLayout is probably not the best point to start. It does not work like all other generators but has this weird two stage process with needs special treatment. It is also pretty complicated. May be you can look at the TypeTray instead or at somethin even simpler.

Obviously this special treatment is missing from the cli code. I'll take a look at it.

PellelNitram, 2019-09-22:
Thank you very much for the information and looking into what is the problem!

I wanted to construct a custom tray with 1x4 elements, hence my start off the TypeTray. Subsequently, I wanted to add some slots for fingers to facilitate the access to 1x4 compartments.

Which generator would you suggest to start off for this purpose? Or would you design it completely from scratch? Apart from how to start the development of the generator, I was a bit scared if the functionality of adding slots for (human) fingers is possible with ``boxes``.

florianfesti, 2019-09-22:
Ok, turned out to be an easy fix: Just emptying the .open() and .close() methods for the first stage.

closed 2019-09-22 by florianfesti

florianfesti, 2019-09-23:
For a 1x4 tray TypeTray should work just fine. TrayLayout is only needed if you need a more complicated layout with some walls being interrupted.

There are two ways to add slots: 
- Use a callback that draws holes. The holes can be adjacent to the border of the piece to generate slots.
- Use a custom Edge class for the border you want slots in. If you want the slots on the inside walls only you can pass an instance of the class to the SlottedEdge instances instead of "e" in line 116 and line 143. If you want the slots on the outside walls you have to replace the simple edges in `t1`, `t2`, `t3`, `t4` by SlottedEdge instances first. Although you don't need (these kind of) slots on those edeges it will split the outsides edges in to small pieces with one separate edge of each compartment.

Have a look at https://florianfesti.github.io/boxes/html/api_examples.html NemaMount is a simple example for callbacks and BinTray an example for replacing the top edges of the TypeTray - even a bit more complicated than what you are planning.

PellelNitram, 2019-09-24:
Thank you very much for the help!

I experimented with both, the callbacks and the custom Edge classes. I focused on the latter, though. The result was unexpected to me but then I realised that I expressed myself incorrectly. Of course the notion of "slot" and "fingers" in my question above can be interpreted as the wooden finger joint sliding into the slot. However, that's not what I meant initially. I meant human fingers.

I attached a figure below which shows what I was able to render with SlottedEdge and what I would like to achieve. The finger is drawn in red.

[image]

Could you maybe give me a hint how to achieve that? I believe that the infrastructure of "callbacks" as well as "custom edges", as implemented in ``boxes``, would be capable of what I would like to achieve. Maybe there is already something implemented in ``boxes``?

florianfesti, 2019-09-24:
Nah, I understood that you were talking about human fingers right away. But may be my answer was a bit confusing.

https://github.com/florianfesti/boxes/blob/master/boxes/generators/notesholder.py uses a similar special Edge class. Adjusting the code a bit should give you the edge as shown above.
For a normal box this would be sufficient. But the TypeTray is more complicated. It uses SlottedEdge which in turn gets passed another edge instance to draw in between the slots. It does this even where there are no slots needed to get those segments at the right places. Right now this is either an "e" (straight edge) or "f" (finger joint) edge. You want to pass an instance of your own class there to get those human finger slots for each compartment.
But TypeTray "cheats" for the edges of the outside walls so you might use SlottedEdges on top of there, too. This is what BinTray has done.

I hope this clears things up a bit.

commit 3ba90c51 (2019-09-22): TrayLayout: Fix cli

TrayLayout was still trying to produce a drawing and failed. Now TrayLayout is
not using most of the infrastructure and only TrayLayout2 has the capacity to
create a drawing.

Resolves: #172