florianfesti/boxes #174: play parameter adds to FingerHoleEdge height

MAKOMO, 2019-09-25:
I defined a custom edge likes the following

```
        U = edges.FingerJointSettings(t, relative=False,
                                      surroundingspaces=1,
                                      space = n*t, finger=m*t, edge_width=o*t,
                                      width=bt)
        u = edges.FingerHoleEdge(self, U)
        u.char = "u"
        self.addPart(u)
```

If I add a play parameter as follows, its value is added to the height of those holes.

```
        U = edges.FingerJointSettings(t, relative=False,
                                      surroundingspaces=1, play=3,
                                      space = n*t, finger=m*t, edge_width=o*t,
                                      width=bt)
        u = edges.FingerHoleEdge(self, U)
        u.char = "u"
        self.addPart(u)
```

Not sure if this is the intended behavior, maybe I am confused about the intension behind this play parameter. 

I think I found the corresponding line in edges.py in the definition of the function __call__ of the class FingerHoles (last two lines).

```
               p = self.settings.play
...
                self.boxes.rectangularHole(pos + 0.5 * f, 0,
                                           f+p, self.settings.width+p)
```

With the third parameter of `rectangularHole` being the dy, maybe the `+p` should be removed here?

florianfesti, 2019-09-27:
Well, the play parameter is an ugly hack. I abused the FingerJointEdge for the slide on lid. And the play allows making the holes a little bit bigger to allow sliding the fingers in and out easily. Default value used there (LidSettings) is 0.05 which is much less than 3. But to reduce friction the holes need to be made bigger in both directions.

What are you actually trying to achieve with your code. May be adjusting the finger and space parameter is the better solution.

MAKOMO, 2019-09-27:
Ok, so I indeed misunderstood the purpose of this parameter. I thought it allows to place fingers on a side with some "tolerance" such that the spacing and the finger size can slightly vary if it helps to fit more on the available space (mostly interesting for larger fingers/spaces).

The intended use of the play parameter makes also sense and it should be promoted from its hack status. I was just looking for make fingers slight easier into their slots (a case where the construction does not depend on a tight fit for stability). I just adjusted the finger and space parameters in the end. In my case it would even make sense to "round" the fingers slightly to make them even easier to stack in. Would be one more parameter...

Very impressed by your lib which I use to make strange things. However, I mostly don't understand its API so I might make things sometimes more complex than they could be...

closed 2019-09-27 by MAKOMO

florianfesti, 2019-09-27:
Yes, the library has some special way of making things easy when done in the right spirit and much more complicated if done otherwise. I tried to convey that in the API docs but it is still a bit hard to get from just there. If you have any particular questions feel free to ask. Improving the docs is much easier when I know they are actually used and helpful.