florianfesti/boxes #177: Custom fingers/edges

Skivmag, 2019-10-09:
I'm looking for a way to redefine standart fingerhole connect with something like on the picture:
[image]
It allows to assemble box without clue
Does it exist simple way to do it?

ZenithDK, 2019-10-10:
Support for this would be awesome. I've bought a number of board game inserts with this kind of finger joints and it works great.

florianfesti, 2019-10-10:
The longer I think about it the easier this seems to be. I'll give it a try the next time when I am at the laser cutter.

florianfesti, 2019-10-11:
Code is not on the web instance yet. Can someone try with a local instance if the result is as it should be, please? In my small test with 3mm it worked just fine. But I am not sure if it will really work with something like acrylic.

closed 2019-10-11 by florianfesti

bferrarese, 2019-10-11:
Trying inkscape extension Misc/AllEdges and I don't see it. I'm guessing the .inx don't have the new option.

bferrarese, 2019-10-11:
Got it working by manually setting 'springs' in my inx

[image]

Skivmag, 2019-10-12:
Today tried on master branch
Works like a charm, thank you!
I'll try it to cut it on Tuesday and will write about results

commit 13ecfc99 (2019-10-11): Support springs for finger joints (FingerJointSettings.style)

Resolves: #177