florianfesti/boxes #178: UnevenHeightBox + TypeTray = Pen Holder

bwente, 2019-10-15:
Not a problem or an issue. Just wanted to share.

[UnevenHeightBox](https://www.festi.info/boxes.py/UnevenHeightBox?FingerJoint_angle=90.0&FingerJoint_surroundingspaces=2.0&FingerJoint_edge_width=1.0&FingerJoint_finger=4.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&bottom_edge=e&x=100.0&y=100.0&outside=0&outside=1&height0=80&height1=80&height2=120&height3=120&lid=0&thickness=3.33&format=svg&tabs=0.0&debug=0&reference=100&burn=0.1&render=0)

[TypeTray](https://www.festi.info/boxes.py/TypeTray?FingerJoint_angle=90.0&FingerJoint_style=rectangular&FingerJoint_surroundingspaces=2&FingerJoint_edge_width=1.0&FingerJoint_finger=4.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&Stackable_angle=60&Stackable_height=2.0&Stackable_holedistance=1.0&Stackable_width=4.0&Hinge_grip_percentage=0&Hinge_outset=0&Hinge_pinwidth=0.5&Hinge_style=outset&Hinge_axle=2&Hinge_grip_length=0&Hinge_hingestrength=1&CabinetHinge_bore=3.2&CabinetHinge_eyes_per_hinge=5&CabinetHinge_hinges=2&CabinetHinge_style=inside&CabinetHinge_eye=1.5&CabinetHinge_play=0.05&CabinetHinge_spacing=2.0&Lid_angle=90.0&Lid_second_pin=0&Lid_second_pin=1&Lid_spring=both&Lid_style=rectangular&Lid_surroundingspaces=2.0&Lid_edge_width=1.0&Lid_finger=3.0&Lid_play=0.05&Lid_space=2.0&Lid_width=1.0&Click_angle=5&Click_bottom_radius=0.1&Click_depth=3.0&RoundedTriangleEdge_height=150.0&RoundedTriangleEdge_r_hole=2.0&RoundedTriangleEdge_radius=30.0&RoundedTriangleEdge_outset=1&sx=25*4&sy=25*4&h=80.0&hi=0.0&outside=0&outside=1&bottom_edge=h&top_edge=e&gripheight=0&gripwidth=0&thickness=3.33&format=svg&tabs=0.0&debug=0&reference=100&burn=0.1&render=0)

Merged in Adobe Illustrator...

[image]

[image]

florianfesti, 2019-10-15:
That's pretty neat! Thanks for sharing!

Note to self: If implementing as generator / option for bottom/edge the slant needs to align with the outer edge of the front/back walls.

florianfesti, 2020-02-01:
Hmmm, what if I make this another edge type. Basically similar to the "h" edge. But there would be 4 instances. So this would require some similar mechanism as the top_edge param where one type is translated to a set of 4 edges that work together. Then this could be added as a bottom for most boxes.