florianfesti/boxes #186: Move to another translation service

florianfesti, 2019-11-01:
Looks like my suspicion about Zanata where justified and the service is not longer properly supported.  So we either need to move somewhere else or have translations edited via GitHub pull requests.

https://hosted.weblate.org seems like a possible solution. 

Do any translators have an opinion/advice on this?

bitbasher, 2023-04-30:
what about google translate .. it can do whole web pages and/or documents, no?

smor, 2024-02-29:
Hello,

I once contributed to the FreeCAD translation project for the French language. This project uses Crowdin to translate .po and .ts files. Here is [the FreeCAD project on Crowdin](https://crowdin.com/project/freecad), and [FreeCAD's localization documentation](https://wiki.freecad.org/Localisation). I receive requests to translate new strings by mail every now and then, when new strings are to be translated, which I believe are discovered via some Git integration. I found it was an overall good experience. 

Crowdin features a [Github integration](https://store.crowdin.com/github) which can push approved translations in a `i10n` branch as a PR, and they are free for FOSS projects. 

Another software I used in the past was Transifex, also free for FOSS projects.

I am not affiliated with any software, but I'm simply interested in contributing to the localization of the software in French !