florianfesti/boxes #187: Noteholder is not working as expected

tschissler, 2019-11-04:
First, I'd like to mention that I was surprised that the behavior of the noteholder box changed completely. In the past I was able to create these kind of boxes which is not possible anymore.
[image]
I'd like to kindly ask to create new types for a change in behaviour.

Second thing is that it seems to have a bug: 
[image]

florianfesti, 2019-11-04:
Yes, the behavior was changed as of #129. The issue above is only showing up for opening == 0.0 which is not really proper use of the generator. But as I added special code for this case let me fix that code quickly.

closed 2019-11-04 by florianfesti

florianfesti, 2019-11-04:
The new design can now be seen in the thumbnail.

commit 4ac46840 (2019-11-04): NotesHolder: Fix various issues

Code for opening == 0.0 was broken (See #187)
Fixed placement of bottom
Fixed margin of USlotEdge

Resolves: #187