florianfesti/boxes #190: Card box generation posibly wrong (Website)

wlma, 2019-11-20:
Hi,
looks like the generation of the cardbox is somewhat wrong. I tried to build a Box, but backside and top doesn't fit...
Backside should imho have also the fingerjoint on top to put the small piece ( german: "Führung") on top. Additionally the cover shouldn't have the fingertips on one side.
(or I'm doing the build completely wrong ;-) )

See: 
https://festi.info/boxes.py/CardBox?FingerJoint_angle=90.0&FingerJoint_style=rectangular&FingerJoint_surroundingspaces=2.0&FingerJoint_edge_width=1.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&h=30&cardwidth=65&cardheight=90&num=2&thickness=3.0&format=svg&tabs=0.0&debug=0&reference=100&burn=0.1&render=1

Apart from that: appreciate all the work done for this great tool, thanks!

florianfesti, 2019-11-20:
Yeah, took me a while to get what's going on here. But the pieces should actually be correct. The thin piece with the finger joint belongs to the finger joint at the lid. It is the grip for the lid. It overhangs the lid to neatly close up the slots holding the lid.

And yes, the thumbnail shows a bit older design that doesn't have this feature yet.

wlma, 2019-11-20:
Ah, this makes sense. Thanks for clarification, Florian.

closed 2019-11-20 by wlma