florianfesti/boxes #192: Possible benefits from removing cairo as dependency

thoka, 2019-11-23:
I've started to experiment getting rid of cairo as dependency. 
It seems, that it is mainly used to create svg, so it should be possible to switch to a pure python solution, for example using [svgwrite](https://pypi.org/project/svgwrite/).

As a test I started [a cairo wrapper]((https://github.com/thoka/boxes/commit/c1e4506a75d39ad1897b490b9d85d9d31a24592d), which creates a cairo and no-cairo svg alltogether, to examine possible problems of this approach.

Creating the svg "barefoot" could have some benefits:
* individual parts could be grouped 
* allows running on windows

To get this running there is some work to do. 
Do you think, it is worth the effort?

florianfesti, 2019-11-23:
I have been thinking about getting rid of cairo for a while tbh. There are many things that are not working great with cairo. One thing that does work great is the ability to also create postscript that can be transformed to many other formats. SVG is kind of a dead end when it comes to converting to other formats.
But having support for SVG only is acceptable IMHO if other formats are still supported when cairo is available. But I doubt that this is really going to work out. Fixing the current issues (non continuous outlines, wrong grouping, ...) can probably not done with the cairo API. So in the long run at least postscript needs to be supported, too. But that should not stop you from going after replacing cairo for SVG at first.

florianfesti, 2019-11-23:
Another thing to consider is whether the cairo API is the best level of abstraction. Yes, right now it is used all over the place. But may be it should be hidden behind a set of methods of the Boxes class. There could then be multiple super classes that have different implementations: One for cairo and one using svgwrite.
This would probably require restricting the use of the cairo context to this super class as a first step.
If this is something to you are interested to pursue I am all too happy to help. Some things like ctx.scale() can probably be hidden behind a very thin façade.

thoka, 2019-11-24:
My thought was: implementing needed parts of cairo in python would open the way to support different formats by its own. It should be possible to do dxf and pdf export quite easily, for example. 

So I'm trying to let the cairo wrapper do all affine transformations, so output in the machine coordinate system can be added in different formats.

thoka, 2019-11-24:
So, its basically working.
Current issues:
* svg post-processor is mixing things up
* text not implemented ( have to find a way to know the font metrics )

I have some questions:
* Why is the current implementation starting with such a high y-offset?
* Why is a white box put around text ?
* Is it important that all output formats create exatly the same text output?
* Currently text alignment is done not by the output generator. Could that be changed?
* Actually there is no concept of parts at the output level. How could that be introduced?

florianfesti, 2019-11-24:
> * Why is the current implementation starting with such a high y-offset?

SVG is using a left handed coordinate system with 0,0 being at the top left as is custom for desktop publishing. Boxes is using a right handed coordinate system with 0, 0 being bottom left as in maths and CAD. The offset keeps all coordinates in positive numbers. But that all not too important. But some SVG readers don't cope well with coordinates not in SVGs positive quadrant

> * Why is a white box put around text ?

My code for calculating page size does not work for text, this is a hack to make sure the text is on the page.

> * Is it important that all output formats create exatly the same text output?

Text is complicated. Cairo generated glyphs and places them on the drawing. SVG probably also supports proper text that will depend on the fonts available when rendering. While rendering the fonts into the file is the saver way for reproducibility I think I am fine with any solution.

> * Currently text alignment is done not by the output generator. Could that be changed?

Sure

> * Actually there is no concept of parts at the output level. How could that be introduced?

For now there is only ctx.stroke for structuring the drawing as that's all that cairo really offers. In the perfect world parts would be in their own SVG group, Part borders would be a continuous lines even if some callback draw details in between the sides. So we'd need new APIs that allows starting and closing new groups and having a stack of open paths that one can return to after closing the most recent one.
Most of the group handling could be done in the move() function which already does similar things for saving the context. Groups would ofc also have a stack for nesting them properly.

thoka, 2019-11-24:
When using a layer bofore the actual output generation, one could collect all drawing primitives in seperate queues and render them in a different order. This would also have the benefit, that boundaries (or in some future the convex enclosure) would be known before final rendering.
I'm not that familiar with the callback machinery to suggest an api for that.
But one could think of some interface, where one can create parts with layers, which all have their own drawing context. I can not oversee, if a part should have a shared cursor/transformation, but many open path queues.

Just for brainstorming: How about connecting this with the color property? 
When switching from rgb values to names (like "boundary", "hole",  with registered translations into rgb) one would switch to a new path and return to the last color/layer with the next restore (with auto-closing, when a color is changed ).

Which generator would you suggest to use as a playground for this experiments?

florianfesti, 2019-11-25:
Collection all the information before writing them out is probably a good idea. I guess cairo does the same thing internally but that's no help for us.

The only thing to know wrt callbacks is that they are executed in between the different outside edges of the parts (like the rectangularWall). So to be able to continue the outside edge we need to be able to start a new path and return to the outside one after the callback is done. There is the cc() helper method that set things up for callbacks and could take care of this path business, too.

Changing the path color will need to open a new path of course but there will be other places that need to manage what path we are currently working on.

May be NemaMount is a good place to play around. It is pretty simple but uses quite some callbacks and even has a standalone hole pattern.

thoka, 2019-12-08:
I've implemented collecting a drawing frist and doing the rendering into svg in a second step (https://github.com/thoka/boxes/commit/22ef3cb76fddf8f776a0766efee0d0567445760d). Fonts still missing. But joining pathes and groups for individual parts work.
[This](https://github.com/florianfesti/boxes/files/3937251/box.zip) is the current rendering of the nema mount (color changes from path to path).
[image]

florianfesti, 2019-12-09:
This looks pretty promising already. I will probably not have time to have a deeper look at the code before the holidays. At first glance it looks straight forward.

Wrt text: It's just fine to render it as a SVG text tag for now. We have to see how well this works with the typical tools. But this is a quick solution to get things going.

florianfesti, 2020-03-12:
Ok, I finally had the time to have a bit deeper look. There are a few minor issues that can probably ironed out quite easily. But there is one big problem: Performance. Rendering a box with a bit flex takes about 3 seconds. This is at least one order of magnitude too slow. A quick run with cProfile shows that is is not the fault of the new code but all happens within `svgwrite` or to be more precise `pyparse` 

```
>>> p.sort_stats(SortKey.CUMULATIVE).print_stats(40)
Thu Mar 12 22:41:58 2020    stats.1

         6524073 function calls (5788127 primitive calls) in 3.258 seconds

   Ordered by: cumulative time
   List reduced from 865 to 40 due to restriction <40>

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
     39/1    0.000    0.000    3.258    3.258 {built-in method builtins.exec}
        1    0.000    0.000    3.258    3.258 <string>:1(<module>)
        1    0.000    0.000    3.258    3.258 __init__.py:610(close)
        1    0.002    0.002    3.227    3.227 drawing.py:273(render_drawing)
        1    0.000    0.000    3.183    3.183 drawing.py:114(save)
        1    0.000    0.000    3.183    3.183 drawing.py:86(write)
        1    0.000    0.000    3.178    3.178 base.py:181(tostring)
      278    0.000    0.000    3.177    0.011 validator2.py:53(check_svg_attribute_value)
        1    0.000    0.000    3.177    3.177 drawing.py:59(get_xml)
     42/1    0.000    0.000    3.177    3.177 base.py:201(get_xml)
       35    0.000    0.000    3.177    0.091 path.py:69(get_xml)
      278    0.000    0.000    3.177    0.011 validator2.py:63(_check_svg_value)
      311    0.000    0.000    3.177    0.010 typechecker.py:307(check)
       77    0.000    0.000    3.177    0.041 validator2.py:43(check_all_svg_attribute_values)
       37    0.000    0.000    3.175    0.086 pyparsing.py:1780(parseString)
       35    0.000    0.000    3.174    0.091 typechecker.py:262(is_path_data)
       35    0.000    0.000    3.174    0.091 svgparser.py:36(has_valid_syntax)
487041/72    0.793    0.000    3.164    0.044 pyparsing.py:1530(_parseNoCache)
  1742/37    0.003    0.000    3.163    0.085 pyparsing.py:4311(parseImpl)
110468/387    0.223    0.000    3.162    0.008 pyparsing.py:3732(parseImpl)
30734/5638    0.093    0.000    3.022    0.001 pyparsing.py:3809(parseImpl)
87804/15526    0.043    0.000    2.782    0.000 pyparsing.py:4096(parseImpl)
 1705/630    0.003    0.000    2.659    0.004 pyparsing.py:4390(parseImpl)
1410/1398    0.005    0.000    2.629    0.002 pyparsing.py:3897(parseImpl)
62200/27048    0.047    0.000    2.539    0.000 pyparsing.py:1613(tryParse)
    71196    0.198    0.000    0.966    0.000 pyparsing.py:4729(postParse)
    65064    0.060    0.000    0.361    0.000 pyparsing.py:4456(parseImpl)
518598/518596    0.317    0.000    0.355    0.000 pyparsing.py:497(__init__)
   518598    0.180    0.000    0.277    0.000 pyparsing.py:488(__new__)
    71196    0.123    0.000    0.255    0.000 pyparsing.py:889(copy)
    71196    0.161    0.000    0.217    0.000 pyparsing.py:558(__delitem__)
    93520    0.077    0.000    0.157    0.000 pyparsing.py:2931(parseImpl)
  1389704    0.136    0.000    0.136    0.000 {built-in method builtins.isinstance}
   164755    0.103    0.000    0.131    0.000 pyparsing.py:795(__iadd__)
    71196    0.087    0.000    0.114    0.000 pyparsing.py:825(_asStringList)
   115361    0.060    0.000    0.060    0.000 {method 'match' of 're.Pattern' objects}
   206536    0.047    0.000    0.051    0.000 pyparsing.py:1511(preParse)
    22804    0.031    0.000    0.049    0.000 pyparsing.py:2736(parseImpl)
   314142    0.046    0.000    0.046    0.000 {built-in method __new__ of type object at 0x7f1aeeec0b40}
    21675    0.021    0.000    0.045    0.000 pyparsing.py:3059(parseImpl)
```

Looks like the lib tries to parse and verify all the path strings but is not very smart about it. For this to move forward we need a way around this.

thoka, 2020-03-13:
should be faster with https://github.com/florianfesti/boxes/commit/886cb7908696de00c11195276463c518bb748e8c

florianfesti, 2020-03-13:
Yes, this fixes the issue.

florianfesti, 2020-03-27:
More news. Good news and bad news...

I have been cleaning up the patches a bit and fixed a few minor issues. In my mind it is very clear that the goal of this must be getting rid of cairo completely. I also want to get rid of all the post processing. I already commented it out and now use code in drawing to move the coordinates.

I left my own new patches pretty small to make them easier to read but I will squash them into bigger patches later on.

I now run into a problem that I was not able to solve right away - even with quite some googling. Applying the transformation matrix (Context._m) to the arc terms of the path does not work. It is fine for most parts but if the matrix is non trivial things go wrong. This can be seen in the SpicesRack generator - probably the only one that uses elliptical curves.

I have not yet check other curves. But they may just work as they parameters are control points which can be transformed easily as all other points.

I pushed my tree to  https://github.com/florianfesti/boxes/tree/nocairo

thoka, 2020-03-27:
Glad to see this progress.

Regarding the arcs: one could add a special command to render the arcs needed for offset correction to still be able to render them as "fast edges".  

All other arcs could be [approximated by bezier curves](https://stackoverflow.com/questions/734076/how-to-best-approximate-a-geometrical-arc-with-a-bezier-curve), which should transform easily (cairo renders all arcs into bezier curves anyway, if I remember correct).

This should be fine for a quarter circle / ellipse.

florianfesti, 2020-03-28:
OK, this did the trick. It also en passant fixed the issues left in the PSSurface.

It broke Path.faster_edges but that was easy to fix.

thoka, 2020-03-28:
Cool. Except for the hacked text output this should be all?

florianfesti, 2020-03-28:
Sort of...

I did quite a bit more clean ups. I got rid of the svgwrite module dependency as we need so little SVG that we can just build the three tags ourselves in pur XML.

The PSSurface still needed some love.

Anyway, things are coming together. I need to sleep at least one night over this - better two and beat the patches into shape. The I can setup a test server. Idea is to have both branches running side by side for a while to iron out the remaining kinks.

florianfesti, 2020-03-29:
OK, squashed most of the patches.
Text placement still needs a bit of love. The PS output could grown a comment with the metadata. But overall this looks pretty good in my book.

I am now considering a DXFSurface but have not yet brought myself to get it started.

florianfesti, 2020-04-06:
Ok, some minor saving-the-world business keeps distracting me. Nevertheless I managed to set up a test instance running the nocairo branch from this repo:

https://www.festi.info/boxes.nocairo

Still need to write an article on hackaday.io to tell people to test.

florianfesti, 2020-04-06:
Current SVGSurface does not yet produce horizontal and vertical lines as did the old cairo post processing. Should be easy to add, though.

florianfesti, 2020-04-13:
OK, the text needs even more work than I thought. E.g. we are not even saving the angle in which it points. This is very noticeable in the DrillBox generator from Misc.

florianfesti, 2020-04-26:
Ok, getting the text right was even more difficult than I expected. I now know much more about PostScript than I ever wanted. Anyway, I updated https://www.festi.info/boxes.nocairo/?language=en to the new code. More testing is welcome.

bferrarese, 2020-04-26:
Hingebox is throwing “float division by zero”.

florianfesti, 2020-04-26:
@bferrarese Thanks for testing! Looks like the code doesn't like arcs of zero degrees. Fixed in db4c8ce52aa9c04c0e57c52076b7d150f7f84c47.

bferrarese, 2020-04-26:
I installed nocairo branch as an inkscape extension and I noticed it now asks for Affine. It works beautifully after installing Affine.

florianfesti, 2020-04-26:
Cool! Yes, there is some clean up work still needed. Like adjusting the dependencies, the install instructions, etc.

florianfesti, 2020-05-07:
OK, moved the main instance over to the new branch to get it more testing. The old back end is still available under https://www.festi.info/boxes.cairo/
I plan to merge the new back end into the master branch on the next week end if there are no big issues popping up - and probably adjusting documentation only after that.

florianfesti, 2020-05-18:
Merged into master. Will update the web instance tomorrow.

There are a few things that can still be improved like actually closing the lines with "Z" and checking for inner corners at the start and end of a closed path. But these are no reason to wait any longer.

florianfesti, 2020-05-19:
OK, I guess I close this here. @thoka : Thanks again for making this happen!

closed 2020-05-19 by florianfesti