florianfesti/boxes #193: Do holes have the correct size?

MAKOMO, 2019-11-29:
Holes are rendered/cutted by the following function. Should the resulting hole in the material have exactly the radius given by r?

If yes, I wonder if the last line should use `_r` instead of `r` to adjust the radius to the given burn/kerv of the laser.

```
    def hole(self, x, y, r=0.0, d=0.0, tabs=0):
        """
        Draw a round hole

        :param x: position
        :param y: postion
        :param r: radius

        """
        print("r",r)

        if not r:
            r = d / 2.0
        if r < self.burn:
            r = self.burn + 1E-9
        r_ = r - self.burn
        self.moveTo(x + r_, y, -90)
        self.corner(-360, r, tabs)
```

florianfesti, 2019-11-29:
Yes, this code looks a bit suspicious. But actually `corner()` does burn correction on its own. So the code looks right to me. The way to check is just increasing the burn and looking at the result.

closed 2019-11-29 by florianfesti

MAKOMO, 2019-11-29:
Ah, ok. I didn't expect the corner to take the burn into account. Makes sense. I will test as you suggested. Sorry for bothering you!

florianfesti, 2019-11-29:
Actually, `corner()` is (nearly) the only place that really takes burn into account. See https://florianfesti.github.io/boxes/html/api_burn.html

MAKOMO, 2019-11-29:
Thanks for this additional information. I still have no broad overview on how that lib exactly works. Too complex for my small brain;)

In the meantime I experimentally confirmed that the hole size is correct.

Have a relaxed weekend,
Marko