florianfesti/boxes #195: python boxesserver error

2ad9krusem, 2019-12-11:
following directions for starting server. Got stuck on second to last step:
"Run the development server with the command python scripts\boxesserver Note: You likely will be notified by your firewall that it blocked network access. If you want to use boxesserver you need to allow connections."

I get error:

C:\Users\Mark\Downloads\boxes-master\boxes-master\scripts>python boxesserver
Traceback (most recent call last):
  File "boxesserver", line 27, in <module>
    import markdown
ModuleNotFoundError: No module named 'markdown'

florianfesti, 2019-12-11:
You are missing the  markdown Python module. This should have been installed in step 7 of the [Windows installation instructions](https://florianfesti.github.io/boxes/html/install/windows.html#native). `pip install Markdown` should fix this. You may need to add `lxml` there, too.

2ad9krusem, 2019-12-12:
The step 7 filenames (from the previously downloaded files’ names in steps 3/4) were changed slightly from the stated directions. I altered the file names to match the new files that were part of download. I then skipped ahead to step 11 and I am still getting the same error.

2ad9krusem, 2019-12-12:
Also… I’m following all of these directions because the box generator program (online) has ceased to function.

florianfesti, 2019-12-14:
Yeah, I though I plugged most of the holes. I plugged a few more now. I hope the server stays up and running.

closed 2019-12-14 by florianfesti

D-Platoon, 2024-02-09:
I know its old, but i need to heat up this thread again.
I have the same problem while installing boxes.py to my local computer (Windows 10).
I installed Python 3.12.2 (latest version), and i think i have done every other steps like in the description.
At Step7 I will also get the following error:
```
D:\boxes-master>python scripts\boxesserver
Traceback (most recent call last):
  File "D:\boxes-master\scripts\boxesserver", line 34, in <module>
    import markdown
ModuleNotFoundError: No module named 'markdown'
```
If I want to reinstall markdown i get the following message:

```
D:\boxes-master>pip3 install Markdown
Requirement already satisfied: Markdown in c:\users\thomas\appdata\local\programs\python\python312\lib\site-packages (3.5.2)
```

lxml is also already installed.

Any ideas what else I can do?