florianfesti/boxes #196: Like to create a "closedbox" with internal shelves or dividers.

st2000, 2019-12-15:
So, reading this:
"  If you want to discuss your idea open a ticket describing it and ask questions there. This is encouraged even if you think you know what you want to do. There are many short cuts in Boxes.py and pointing you in the right direction may save you a lot of work  "

I would like to create a new "closedbox" with any number of internal (hidden) shelves or dividers.  I want to incorporate a finger-joint approach to internal shelves to mount electronic components on.  I looked at the "closedbox" python script.  But was confused as to how the rectangularWall() function works. 

Guessing, perhaps adding a line such as this will add an internal shelf to the box:
        self.rectangularWall(x, y, "ffff", bedBolts=[d2, d3, d2, d3], move="mddle")
...but more likely there is no 'move="middle"' like directive.

florianfesti, 2019-12-15:
You may want to have a look at the TypeTray generator. You can select a "f Finger joint" as top_edge to get a closed box.

Wrt rectangularWall: You can leave out the bedBolts param. The move parameter only deals with arranging the part on the drawing. To place a wall inside the back you need to punt fingerHolesAt the right place. You can use callbacks for doing this. Details are explained [here](https://florianfesti.github.io/boxes/html/api_parts.html)

st2000, 2019-12-15:
Sorry, I'm very new to this.  Starting with the typetray generator, when you say: " You can select a "f Finger joint" as top_edge to get a closed box. ".  I'm confused as it looks like the TopEdge is already using a finger joint:
`        self.addTopEdgeSettings(fingerjoint={"surroundingspaces": 0.5},`

For the balance of the changes (my goal is a box with n number of parallel shelves) I assume I need to remove the some inner walls (last 9 lines of code) and various call backs associated with holes for the removed 2 inner walls.

But what of adding finger joints to the 4th side of the 2 remaining inner walls as well as all the outer wall edges that are currently smooth?

florianfesti, 2019-12-15:
Sorry, I should be more clear: The TypeTray generator is already able to do this as is.
At https://festi.info/boxes.py/TypeTray?language=en there is a top_edge parameter you can just change to "f Finger joint". If you don't want wall in one directions you can just change sx or sy to a simple number.

Yes, the TypeTray generator could use a bit of documentation...

st2000, 2019-12-15:
Oh!  I get it, the Type Tray generator is more like a general box generator which can create the box I described above.  Very good.  Thanks!

st2000, 2020-07-11:
Pardon me for reopening this issue rather than creating a new one.  But I really want to do the same (add an internal shelf) to a different box and thought it simpler to continue with the same issue rather than explain my self all over again...

I would like to add an internal shelf to the "RoundedBox" design.  I'm looking at the RoundedBox Python script and was hoping I could see my way to adding an additional top to the design and place it somewhere in the middle of the box.  But I just don't see it.

st2000, 2020-07-19:
Asking question in issue #229 so closing this issue.

closed 2020-07-19 by st2000

florianfesti, 2020-07-20:
To answer the question about the 3rd roundedPlate() call: It is passed `move="up only"` the means the current position is moved by the high of the part without actually drawing it. This places the surroundingWall() above the one or two actual .roundedPlate()s