florianfesti/boxes #197: Color preferences for different CAM software

tanah, 2019-12-17:
Is there a proper way to change the line colors without hacking the Color class? The CAM program I'm using (K40 whisperer) uses red for cutting, black for rastering and blue for vector engraving.

florianfesti, 2019-12-18:
Well, we discussed this issue when introducing the svg_Ponoko format in #105 and #112. The output format and the style used (colors, line width, ...) should actually be two separate settings that can be combined freely. But this would just clutter the UI and not all cobinations would make sense anyway. So instead we just added the svg_Ponoko as one more format - even if it doesn't really fit in there: c5d434105b3cd69605b9037435b68e35767d5080

To do this properly we'd need some "style" classes that translate semantic tags like outline, hole, text to colors and line styles. Maybe for the UI we could just add just another svg_* format while having the split on the code level only.

neophrema, 2021-06-18:
Im sorry to ask, did anything in this subject changed meanwhile? I just stumbled over the same issue and prolly need to script some color substitution script.
Nevertheless I think setting the Parameters in commandline might be super useful.

florianfesti, 2021-06-19:
Not really. There has been a little bit of progress with 9a46882c8fb647a86b51d9f93135a6b661960190. Some good idea on how to do the user interface would help. I have the sneaking suspicion that this requires the use of cookies to really work well. Then there is the question if you want to offer an interface to choose your own colors or just offer a list of predefined sets.

thoka, 2023-02-14:
Hm. How about reading (complex) configuration values from a JSON / YAML file somewhere on the web, provided by one URL option?
This way people could develop settings in their git repos and one would omit the need for a user interface.