florianfesti/boxes #203: Flexbox finger joints

bwente, 2020-01-30:
I am trying to make something like this...

[image]

Is there a generator that will get me close? I made this one, but would like an easier way to adjust for the thickness of the material.

florianfesti, 2020-02-01:
Hi!
The [Rounded Box](https://festi.info/boxes.py/RoundedBox?language=en) already is pretty close.
There already is [.surroundingWall()](https://github.com/florianfesti/boxes/blob/master/boxes/__init__.py#L1558) which basically is the part you need. So you might consider extending the [.roundedPlate()](https://github.com/florianfesti/boxes/blob/master/boxes/__init__.py#L1485) which is used for top and bottom or implement a new part.

You may want to switch the edges on top and bottom for the surroundingWall from "F" to "f" (and vice versa for the roundedPlate or its replacement. This will require additional work as those pieces will not yet handle the steps needed between the edge and the flex part ("X" edge) which still needs to stick out to be supported around the corner.

Depending on what you feel comfortable with you can either just clone the [Rounded Box generator](https://github.com/florianfesti/boxes/blob/master/boxes/generators/roundedbox.py) or if you feel comfortable with that extend it in place.

bwente, 2020-02-01:
Thanks! My comfort level is randomly changing values on your site until I get something close ;)

I used these settings...
https://www.festi.info/boxes.py/RoundedBox?FingerJoint_angle=90.0&FingerJoint_style=rectangular&FingerJoint_surroundingspaces=1.0&FingerJoint_edge_width=12&FingerJoint_finger=4.0&FingerJoint_play=0.0&FingerJoint_space=4.0&FingerJoint_width=12.0&x=120.0&y=120.0&h=120.0&outside=0&outside=1&radius=21&wallpieces=1&top=closed&thickness=3.5&format=svg&tabs=0.0&debug=0&reference=100&burn=0.1&render=1

(thanks again for adding the render parameter)

I used that file and manipulated it in illustrator to get the image below.

The green is boxes.py, the red is what I added to get the results in blue.
[image]

florianfesti, 2020-02-01:
Looks pretty good already! I'd be interested in some feed back how well it works an whether those long slits at the corners weaken the parts too much or not.

bwente, 2020-02-01:
I think it worked perfectly!

[image]

bwente, 2020-02-01:
Here is the completed results. I did not sand or stain it yet.

[image]

florianfesti, 2020-02-01:
Looks really nice. Damn! Now I need to program this...

bwente, 2020-02-02:
Next one I will need to add click on back (bottom).

[image]

closed 2020-02-02 by bwente

florianfesti, 2020-02-28:
https://festi.info/boxes.py/RoundedBox?FingerJoint_angle=90.0&FingerJoint_style=springs&FingerJoint_surroundingspaces=2.0&FingerJoint_edge_width=1.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&x=100.0&y=100.0&h=100.0&outside=0&outside=1&radius=15&wallpieces=1&edge_style=h&top=lid&thickness=3.0&format=svg&tabs=0.0&debug=0&reference=100&burn=0.1&render=1

While not exactly the Click-on edge the new FingerJoints with style : "springs" may do a similar job.

Anyway: Thanks for the inspiration!

florianfesti, 2020-03-01:
Made this ticket into a Hackaday.io log: https://hackaday.io/project/10649/log/174365-a-boxespy-story