florianfesti/boxes #205: Home Design

christensonb, 2020-02-23:
I was working on creating a home design modeling library, when I ran across this library which I think looks perfect  for doing the backend of this project.

My basic is to take a ascii text file (see [floor1.txt](https://github.com/florianfesti/boxes/files/4241195/floor1.txt) of the layout of a house, and from that plus a table file defining height of walls and windows to generate a SVG file for glowforge to cut out the wall to create a miniature model of the home.

I have forked this repo am beginning to refactor my code to fit within this repo's design.
Any suggestions would be appreciated.

christensonb, 2020-02-23:
My first question would it be possible to have an argument of a path to a file within the web UI.

christensonb, 2020-02-23:
Notes: 
- To create the walls for the room a custom generator will be used, similar to unevenheightboxes:
https://www.festi.info/boxes.py/UnevenHeightBox?language=en
-

florianfesti, 2020-02-23:
This are a lot of questions, let me see if I can come up with answers:
1. Sure. That just need a class similar to ArgparseEdgeType or BoolArg. It will probably be even more involved as it needs to get the file data from different source depending whether it is used on the cli or in the web instance.
2. This is basically just because I am much too old school. PRs are welcome. Note that there are some docker file that use the same patterns (In case you want to fix them, too. Switching one thing after the other is also fine).
3. Including it in pypy sounds interesting. I am not too keen on `festi_boxes`. I need to look into this to find a good name (good names are hard)
4. Currently there is only one output file. Adding support for multiple files (e.g. basename.svg, basename-part2.svg, ...) should not be too hard. But this will need manual work from the generator. Currently the whole placing of arts is done manually - and typically not very smart. This is kinda on purpose as things change with different sizes and people need to adjust for the size of the material used anyway.
Breaking up single parts is a whole topic of its own. Not sure if we want to get into this right now.

florianfesti, 2020-02-23:
You may want to consider placing measurements into your file formats. While I understand you want to have your ASCII plans look to scale, they are hard to edit. Doesn't have to be as ugly as TrayLayout (which is kinda similar). You could put the sizes into the walls.

christensonb, 2020-02-23:
1. Ok I will need help with that, but probably after my first real PR
2. Ok I might do this someday, but not sure when.
3. I can do the pypy part, but no rush from my side
4. This assumes though I could get it all done on one sheet which I am pretty sure is not the case.
  That being said I was planning on having a --room-filter because I didn't want to have to cut out the whole thing again just because I changed two rooms, so I guess that can work for now.
5. so I need to have a 2:1 ratio for characters for it to look right, but I will consider this in later versions.

christensonb, 2020-02-23:
Are you okay with this concept of ``layout``?
https://github.com/SeabornGames/boxes/tree/model_home

closed 2023-01-05 by florianfesti