florianfesti/boxes #208: generate.sh has reference to missing box generator

christensonb, 2020-02-23:
```
+ ../scripts/boxes box --x=50 --y=50 --h=70 --output=box.svg
Unknown generator 'box'. Use boxes --list to get a list of available commands.
```

christensonb, 2020-02-23:
--version doesn't work
--help only works if a <generator> is supplied

florianfesti, 2020-02-23:
Yeah, this has experienced quite some bit rot. May be the script needs to be redone to just execute `--list` and then run all generators. There was the idea to make this into a test suite. But that never really happened. One issue is that the resulting file change easily.
Not really sure what to do this this right now...

florianfesti, 2020-02-23:
Still thanks for pointing out that *something* should be done about this.

christensonb, 2020-02-23:
Even still I like this code base ALOT, thanks for writing it.

commit 416f9a8a (2020-03-07): examples/generate.sh: Remove no longer existing generator "box"

Thanks to Ben Christenson for pointing this out.

Resolves: #208
closed 2020-03-07 by florianfesti