florianfesti/boxes #209: Card box lid problem

myk3y, 2020-02-26:
I could be doing something wrong, but the lid for every card box I've made ends up with tabs on one side, where it should be a plain edge on all four sides.

florianfesti, 2020-02-26:
Yeah, the picture is a bit outdated. The tabs there are actually intentional. There is a narrow piece that fits in there that acts as a grip to be able to open the lid more easily.
Guess I finally need to add some assembly instructions as you are already the second person to get confused about this.

myk3y, 2020-02-26:
Thanks Florian.

Makes sense now = doh! :)

myk3y, 2020-02-26:
User error

closed 2020-02-26 by myk3y

bwente, 2020-05-23:
I thought I had done something wrong also. I came back to double check the photo to see how to put it together. I got it figured out after reading these issues.

Here is my final assembly, maybe it will help the next person.

[image]

[image]

This was a gift for my brother-in-law.

florianfesti, 2020-05-26:
Added as baf996764ca0dc5bff956839c913a66b3ca77967 (web instance not yet updated)
@bwente I hope just adding the picture to the repo is ok with you.