florianfesti/boxes #212: Extensions grayed out and unusable in Inkscape on Mac OSX Catalina

dubchild, 2020-03-04:
I finally got the extension menu installed without errors and it shows up but they are all GRAY. I have updated everything and jumped through a million hoops. I have read all there is to read on this script and I am about to throw in the towel because I've wasted so much time on it. It would be really helpful to make a quick video of installation on all platforms because this thing is quite needy and fragile. 

I have spent more than 6 hours trying to get this script to work in inkscape. The web version is great but the promise of running it locally inside inkscape was enough to make me jump in. Please help.

florianfesti, 2020-03-04:
They are greyed out then the .inx files are found but the boxes script is not. Place a copy or a symlink into the extensions directory where the .inx files are.

dubchild, 2020-03-04:
Thanks for your response. However, as you can see in the screenshot, the boxes script is already right there with the .inx files and it doesn't work.

[image]

florianfesti, 2020-03-04:
Can you try removing the following line from one of the inx files:

`<dependency type="executable" location="path">boxes</dependency>`

dubchild, 2020-03-04:
I removed that line from boxes.py.AllEdges.inx and restarted inkscape. Still grayed out. Am I supposed to do something else?

florianfesti, 2020-03-04:
Hmm...
Ok, as we are desparate: Try to also remove

```
  <script>
    <command reldir="extensions">boxes</command>
  </script>
```
at the bottom of the file. If this still doesn't show a menu you are editing the wrong file.
Ofc this will make the plugin not do anything but we need the UI to show up first.

dubchild, 2020-03-04:
Nothing changed. I'm editing the files at Inkscape.app>Contents>Resources>share>inkscape>extensions. If not those files, which path should I be looking at?

florianfesti, 2020-03-04:
`inkscape -x` gives the path inkscape looks for extensions.

jandion, 2020-03-05:
I just had the same issue, I managed to fix it by adding the directory where I have the boxes file to the path system variable. It worked on Windows 10, but I guess it will be the same on mac or linux

dubchild, 2020-03-05:
Yeah I tried that on my macbook and it didn't change anything. Maybe i'm doing it wrong but i added it to my path and still no love. I guess i'm just going to run the local server and load the output manually which is what I was trying to avoid doing. Inkscape is a huge turn-off anyway. I have Illustrator and Affinity which I'm much more comfortable with. 

Boxes.py is great, but the Inkscape plugin setup definitely needs some work. Only the most technical and dedicated users will ever figure it out. I had 2 other professional coders look at it and they couldn't figure it out either. I understand it's free software, but at the same time I'm really disappointed.

Moini, 2020-05-22:
Here's the documentation for updating Inkscape extensions for Inkscape 1.0:
https://wiki.inkscape.org/wiki/index.php?title=Updating_your_Extension_for_1.0

For remaining questions regarding how to update, visit https://chat.inkscape.org/channel/inkscape_extensions .

jgforbes, 2020-12-11:
Any resolution for the grayed out issue?
I am getting the same thing on  Fedora 33 with a fresh installation of inkscape and latest version of boxes.py.

chill117, 2021-02-18:
So I ran into the same problem described in this issue. After a lot of trial and error, I finally got it working by:
* Copying the `.inx` files located at `./inkex/*.inx` to `~/.config/inkscape/extensions/`
* Copying the boxes script located at `./scripts/boxes` to ~/.config/inkscape/extensions/`
* Adding the inkscape extensions directory to my user's path - e.g:
```bash
echo $'\nexport PATH=$PATH:~/.config/inkscape/extensions' >> ~/.profile && source ~/.profile
```

neophrema, 2021-06-18:
Im not sure, I followed what you suggest, I still get kind of an error, and its still grayed out, not sure if this is related:

```
** (org.inkscape.Inkscape:5114): WARNING **: 13:53:58.166: Image file ('/home/plato/.config/inkscape/extensions/./WineRack-thumb.jpg') not found for image widget in extension 'info.festi.boxes.py.WineRack'.
Shortcut::read: invalid verb: DialogSpellcheck
```

florianfesti, 2023-04-01:
boxes_proxy.py got added to the repo a while a go. So this should work now. Please reopen if things are still broken.

closed 2023-04-01 by florianfesti