florianfesti/boxes #217: Three and Four Join edge

christensonb, 2020-04-06:
[image]

I think a really cool edge would be a wide dovetail that has holes like
[image]

if we make the fingers with a slight  widening and we put a slit  from the hole to the edge of the dove tail.  Then as you push the fingers in it would make the dove tail joint even tighter.

I will draw something up to show my intent

christensonb, 2020-04-06:
[image]

christensonb, 2020-04-06:
So this would allow for a three piece joint, and to do the four piece joint I would simply do the three piece every other hole so each of them share the holes.

christensonb, 2020-04-07:
I created some code and it is looking pretty good but I definitely could use some help.
https://github.com/SeabornGames/boxes/blob/duckbill_joint/boxes/edges.py#L1836

It currently doesn't work for even number of tabs, and the DuckbillJointCountnerPart doesn't draw the holes for reasons I can't figure out.

The whole thing probably doesn't scale.

christensonb, 2020-04-07:
[Archive.zip](https://github.com/florianfesti/boxes/files/4442081/Archive.zip)
[image]

florianfesti, 2020-04-07:
I am not quite sure if this solves the right problem (What ever this means). Dovetail joints are only really needed if you bend walls around some object like in RoundedBox. For all other cases you can just make the wall from one piece.

The issue is jointing the other two walls at the some spot - which is something this solution doesn't provide. But may be I am just not getting the problem you are trying to solve.

What about having a FingerHole pattern that allows inserting walls from both sides?

christensonb, 2020-04-07:
I couldn't follow the clockwise counter clockwise issue, but is that why the holes doesn't showup on the right side

christensonb, 2020-04-07:
So if I number the image above from top left to bottom right
1 2
3 4
5 6

Then I could connect 1, 2 and 5 together to make a T intesect.
Alternatively I could connect 3, 4, 5, 6 together to form a +

The problem with assuming you could build it with one long wall is that my glowforge laser printer is only so big.

christensonb, 2020-04-07:
remember 2 is missing the holes that 5 has, and that is a problem I am asking help with.

christensonb, 2020-04-08:
I have mostly solved it
[image]

Do I have to use the radius?

christensonb, 2020-04-08:
https://lh3.googleusercontent.com/-9koQB6EZn4U/Xo00DWdg-2I/AAAAAAAAq8Q/_AC1AzIhXGkiOR4KI3bRmfkga5hGhr34ACK8BGAsYHg/s0/2020-04-07

christensonb, 2020-04-08:
[image]

christensonb, 2020-04-08:
[image]

christensonb, 2020-04-08:
[image]

florianfesti, 2020-04-08:
The joint would be even stronger if you made the holes not part of the outline. Just put some rectagularHole() there. You can even do that directly before drawing the straight line on the outside of the dove tail to save you the math.

florianfesti, 2021-06-05:
I am still kinda missing a good use case. I really don't see why the dovetail joint would be needed there. May be doing less is more and just have a `-|-` joint is more useful.

Closing this for now.

closed 2021-06-05 by florianfesti