florianfesti/boxes #221: Error when click on Electronic Box. Python 3.7. MacOS Mojave.

rhapsodyv, 2020-04-20:
Error when click on Electronic Box. Python 3.7. MacOS Mojave.

`
Traceback (most recent call last):
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/wsgiref/handlers.py", line 137, in run
    self.result = application(self.environ, self.start_response)
  File "./scripts/boxesserver", line 464, in serve
    return self.args2html(name, box, lang, "./" + name, defaults=defaults)
  File "./scripts/boxesserver", line 248, in args2html
    result.append(self.arg2html(a, prefix, defaults, _))
  File "./scripts/boxesserver", line 164, in arg2html
    input = a.type.html(name, default or a.default, _)
TypeError: html() takes 3 positional arguments but 4 were given
127.0.0.1 - - [20/Apr/2020 11:21:00] "GET /ElectronicsBox?language=en HTTP/1.1" 500 59
`

florianfesti, 2020-04-20:
This looks like the new boxesserver is running on an old boxes.py library. Yes, the html() method was currently changed to get an gettext instance passed as 4th param. But here EletronicsBox works just fine.
I double checked and committed 9ea1d55a9403c107deafd073f2bbb8a5d94ed773 but that should not affect your issue.

rhapsodyv, 2020-04-20:
Didn't work.

Seems the only fix is changing:
input = a.type.html(name, default or a.default, _)
to
input = a.type.html(name, default or a.default)

I know python, but not very deep to know how works named arguments... but changing this way fixes the problem and the form shows up.... 

But I get another error when submit the form:

Traceback (most recent call last):
  File "./scripts/boxesserver", line 488, in serve
    box.metadata["url"] = self.getURL(environ)
AttributeError: 'ElectronicsBox' object has no attribute 'metadata'

rhapsodyv, 2020-04-20:
Maybe the problem is with my python version?

florianfesti, 2020-04-20:
You clearly have some old version of Boxes.py installed somewhere on you system that is earlier in the Python path. metadata was added to the Boxes class last summer:

```
fff74cf3 boxes/__init__.py (Florian Festi     2019-07-06 11:47:41 +0200  247)         self.metadata = {
fff74cf3 boxes/__init__.py (Florian Festi     2019-07-06 11:47:41 +0200  248)             "name" : self.__class__.__name__,
0178a853 boxes/__init__.py (Florian Festi     2019-07-04 21:43:56 +0200  249)             "short_description" : self.__doc__,
fff74cf3 boxes/__init__.py (Florian Festi     2019-07-06 11:47:41 +0200  250)             "description" : self.description,
fff74cf3 boxes/__init__.py (Florian Festi     2019-07-06 11:47:41 +0200  251)             "group" : self.ui_group,
fff74cf3 boxes/__init__.py (Florian Festi     2019-07-06 11:47:41 +0200  252)             "url" : "",
fff74cf3 boxes/__init__.py (Florian Festi     2019-07-06 11:47:41 +0200  253)             "command_line" : ""
fff74cf3 boxes/__init__.py (Florian Festi     2019-07-06 11:47:41 +0200  254)         }
```

rhapsodyv, 2020-04-20:
Sorry. You are right. I really don't remember the old install. I was using from a clone, but had some old install...

closed 2020-04-20 by rhapsodyv