florianfesti/boxes #222: 5 Sided 'lcd display' case - how?

QuitButton, 2020-04-27:
I'm trying to make a case to hold electronics and a TFT display, something similar to this shape

https://www.thingiverse.com/thing:2971076

Where the front is angled for the display. I can't seem to get anywhere near that shape - am I missing something?

florianfesti, 2020-04-27:
Yes, this is currently missing. Someone(tm) still needs to program this. If you are in a hurry you can try to abuse the [RegularBox](https://www.festi.info/boxes.py/RegularBox?FingerJoint_angle=90.0&FingerJoint_style=rectangular&FingerJoint_surroundingspaces=2.0&FingerJoint_edge_width=1.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&h=100.0&outside=0&outside=1&radius=50.0&n=5&top=angled+hole&thickness=3.0&format=svg&tabs=0.0&debug=0&reference=100&burn=0.1&render=1). That's not quite what you want but may just work.

florianfesti, 2020-04-28:
Here a simple version: https://www.festi.info/boxes.py/Console?language=en
in commit 279b60abc85574958c990569fb8737d2b9ac5140

I might add a bit more refined one later. With feet and an panel that extends a bit over the sides and walls.

QuitButton, 2020-04-29:
That is brilliant, thank you for adding that, it'll be very useful.

There's one issue I've noticed already, when increasing the X size the top isn't sized up accordingly, for example setting X to 150 results in the attached image
[image]

florianfesti, 2020-04-29:
Yeah, I actually woke up this morning thinking I messed up x and y. But then I thought: "Nah, It'll be fine. Doesn't matter if x goes back to front."
Anyway, fixed in 54d4b6bebdfc212043b4eb1c0211865a4212d19b

florianfesti, 2020-04-30:
Ok. I made some progress on a bit more refined variant yesterday:
[image]

I should be able to get this online the upcoming week end. For now there is still some method missing to attach the front panel. Should it be glued and left without holes or is it better to get attached with some finger joints? I am also pondering whether there should be (another) door or removable wall. The floor or the back wall could get screw holes similar to the lid of the ElectronicsBox. Any opinions on that?

QuitButton, 2020-04-30:
That is looking fantastic. 

I'm guessing most would want the a removable front without tabs, but could it (or optionally the top) be attached by the triangle corner mounts or similar like the lid on the ElectronicsBox? 

And yes, it would nice if either the back or floor could be made to have the outside mounts like the base on the ElectronicsBox.

QuitButton, 2020-04-30:
..and having made that comment I've just realised that the sides attached to the edge of the base so that idea won't work.

florianfesti, 2020-05-01:
Ok, I now have the back wall and the panel removable. The generator is still untested and will most likely need a few more tweaks. So it is currently still in "Unstable": https://www.festi.info/boxes.py/Console2
I also still need to write some assembling instructions as it is a bit of a 3D jigsaw puzzle right now.

florianfesti, 2020-05-02:
Turns out Console still had some issues: #224

florianfesti, 2020-05-02:
OK, a did several changes to make the Console2 actually work. There two small latches that holds the back wall at the top (6 small parts on the bottom right). They lock into place when closed and need to be pressed in to be re-opened. The front panle now has a frame underneath which is held in place with four spring loaded tabs (6 parts at the top right). The idea is that the panel is glued to this frame to have a clean surface.
I am currently thinking about replacing the latches by the same type of tabs. But I might be just a little bit too proud of the latches to do so...

QuitButton, 2020-05-05:
Nice :) It does need a photo showing how these latches work as I can't work it out from the drawing... but that's not unusual for me.

florianfesti, 2020-05-05:
There are now pictures and a description on the bottom of the settings page.

QuitButton, 2020-05-05:
That lid and its fixings are something else 👍  Love it 🥇

florianfesti, 2020-05-19:
OK, generators are now in "Boxes". I also added SideDoorHousing which is a rectangular box with one or two removable walls similar to the back wall of Console2.
Guess this is all done. Closing. Thanks for pushing this and for your feedback!

closed 2020-05-19 by florianfesti

florianfesti, 2020-05-22:
Added a project log: https://hackaday.io/project/10649/log/177879-new-boxes-for-your-electronics-projects