florianfesti/boxes #223: universal box single hinge eye (option i) lid width issue

pinbot, 2020-04-30:
The example pictures show a universal box with a single lid and hinge eye attachment and the width of the lid is the same as the width of the 'pins'  meaning there's basically a cutout for the eye.  But when option i is selected the lid that is generated is narrower everywhere, so the the pin just sticks out from an otherwise uniform edge.  This means the lid doesn't actually cover the sides of the box.  If option k is selected (2 pairs of eyes) then the two lids generated are both full width with a notch for the eye, like the picture.  I can't find any option to affect this.  I tried to be more helpful than this and spent some time looking at at drawLid, but I'm not even clear on how it is drawing the pins let alone how it is falling to do a cutout.  But I certainly note there's a lot more going on for the k option than there is for the i option.

pinbot, 2020-05-02:
3000 more words since I think I struggled to describe it (and didn't notice inline images are supported) Here's the doc image showing the option i lid full width with notch:
[image]
But here's how option i comes out--narrow lid instead of notch:
[image]
Option k does have the full width, but of course has two lids and a center opening:
[image]

florianfesti, 2020-05-02:
Checking the `outset` box in `Settings for Hinges and HingePins` works for me. But yes, the description of he settings and also the defaults could be better.

pinbot, 2020-05-04:
Argh...I mean thank you!  Yes it works for me too.  I thought I'd tried everything AND I thought I'd confirmed there was nothing in the code to do it--fail on both, lol.  Thanks again.

closed 2020-05-04 by pinbot

florianfesti, 2020-05-05:
OK, I changed the default to True, as this is the variant that most people will want.