florianfesti/boxes #224: Problems with box of console type generation

st-stalker, 2020-05-01:
Today i tried to cut console [here's setup](https://www.festi.info/boxes.py/Console?FingerJoint_angle=90.0&FingerJoint_style=rectangular&FingerJoint_surroundingspaces=0.5&FingerJoint_edge_width=1.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&Stackable_angle=60&Stackable_height=2.0&Stackable_holedistance=1.0&Stackable_width=4.0&x=100&y=120&h=100&hi=50&angle=60&thickness=4.0&format=svg&tabs=0.0&debug=0&reference=100&burn=0.1&render=1) - but in the end i couldn't fit back panel - the tabs was overlapping. I can't find a way where I could make mistake, so i assume, that there could be mistake in the code.

Here's photos of the result: [image]
[image]

I wanted to make more cool photos to submit, but... seems, I'll do it next time.

florianfesti, 2020-05-01:
Yeah, there is a bug. For uneven number of walls the parts can't just all have the same edge on both sides. Give me a few minutes and I'll try to fix that.

florianfesti, 2020-05-01:
Ok, fixed in 949c8d8324b1454b7f77a1a78dd106064238a755. It should be enough to just re-cut the back wall as everything else is unchanged (right-most part).
 Sorry for the inconvenience!

florianfesti, 2020-05-01:
Yeah, making good photos is tricky especially of transparent pieces like this. Having a good picture for the generator would be really great.

florianfesti, 2020-05-16:
OK, closing this ticket as everything seems fine now. If you still have some good photos you can still attach them here or open a new ticket or PR.

closed 2020-05-16 by florianfesti