florianfesti/boxes #226: Why are the fingers not rectangular in the corners

Rasputin00, 2020-06-02:
Hi everybody,
first of all, I would like to say you how great your box creators are! Thanks to everybody for sharing.
I read a lot on the how to section and all the stuff, but there is one thing which makes me confuse.

I managed to create a box (the hingebox)  with the online creator. 
It works, but if I zoom in to the fingers, I can see the corners aren't recangular. There is somesting like a microsopic loop.
Is there a purpose behind of this fact?

Would be nice if somebode could give me a feedback to this issue.
Watch the picture...it is yellow marked..
[image]

I can fix it manual but it needs a lot of time....;(
best regards and thanks in advance..
dani

florianfesti, 2020-06-02:
Yes, this is on purpose. See https://florianfesti.github.io/boxes/html/api_burn.html for details. Note that the documentation is a bit outdated as since a few days most inner corners use a better variant which avoids changing direction.
If you do not want those artifacts you can set `burn` to zero. But you then have to do your kerf correction yourself.

Rasputin00, 2020-06-02:
Hi,
thank you very much for helping so fast. Absolutley fantastic!
Now I understand how it works. Nice solution for a problem! I asked myself how the problem with the changeing lenth caused thrue the different material is solved and now you gave me the solution...
What a great idea.
Have nice day....

best regards
dani

closed 2020-06-02 by Rasputin00