florianfesti/boxes #229: Likely a simple question: Changing a local copy roundedbox.py results in indentical SVG patterns.

st2000, 2020-07-19:
Hi,

This is likely a simple question that many can answer.  I created a local git copy of this project and started in on trying to understand how the roundedbox.py Python script worked.  I started by deleting a few lines trying to determine how they affected the resulting SVG cut pattern.  To my surprise the changes didn't do anything.  At first I thought this was because I didn't restart the server.  But I've been restarting the sever after every edit to the roundedbox.py file.  And still all the resulting SVG files are the same.  What am I not seeing?

-thanks

florianfesti, 2020-07-19:
Guess the boxesserver script uses another (probably the system's) installation of boxes.py. Try setting the PYTHONPATH environment variable or change the try except block in boxesserver that imports boxes.generators to look for the local copy first.

st2000, 2020-07-19:
I think that is a good suggestion @florianfesti.  However, when commenting out code I did introduce a syntax error and boxserver called me on it.  So, either I know so little about Python that I don't know what to remove to make a change (I find that hard to believe as I finally removed all the Python code from boxes/generators/roundedbox.py leaving only the header comment in frustration) or I'm not seeing the forest for the trees.

Now I'm starting to wonder if I'm looking at the correct Python source code. 

I want to alter this design:
http://localhost:8000/RoundedBox?language=en
...as served up by this instance of boxserver I run on my computer:
~/git/github/boxes/scripts/boxesserver
...by understanding and editing this Python source code in my git copy of the project on my computer:
~/git/github/boxes/boxes/generators/roundedbox.py

st2000, 2020-07-19:
Follow up...
I think there is some caching going on WRT my web browser which made seeing changes in the Python script difficult.  I now load the web page in twice to make sure I am using the current copy of Python script (why shift+load doesn't do this dependably is a question for another time).

My goal is to add an additional shelf to the roundedbox.py design (I would like to build a Word Clock based on this box).  @florianfesti, do you have any suggestions as to how to do this?

florianfesti, 2020-07-19:
Not quite sure if I understand what you want to do. If you want to have another horizontal wall you need another roundedPlate() and you need callbacks for the .surroundingWall that places the .fingerHolesAt() all the right positions.

See https://florianfesti.github.io/boxes/html/api_parts.html#the-callback-parameter for an explanation of the callbacks.

st2000, 2020-07-21:
Thank you @florianfesti .  Adding an additional wall was easy.  But adding the fingerHolesAt() is still a mystery to me.  I commented out what I think is the correct line of code to modify and added my version below.  As there appears to be no syntax errors I assume the code interprets cleanly.  But I am guessing at the fingerHoldsAt() parameters:

```
#        self.surroundingWall(x, y, r, h, pe, pe, pieces=self.wallpieces)
        self.surroundingWall(x, y, r, h, pe, pe, pieces=self.wallpieces,
                             callback=[self.fingerHolesAt(2, 2, self.thickness)])
```

st2000, 2020-07-22:
So, by searching though the project's Python code I found enough examples to put together the following code which appears to render finger holes in the correct position in the side of the box:

```
#        self.surroundingWall(x, y, r, h, pe, pe, pieces=self.wallpieces)
        y_here = y/2
#        y_here = 4
        x_here = 57.5
        x_space = 86.5
        self.surroundingWall(x, y, r, h, pe, pe, pieces=self.wallpieces,
                             callback=[
                                 self.fingerHolesAt(x_here, y_here, 60, angle=0),
                                 self.fingerHolesAt(x_here + x_space, y_here, 60, angle=0),
                                 self.fingerHolesAt(x_here + x_space + x_space, y_here, 60, angle=0),
                                 ])
```
...however, this will likely break as soon as I add length to the box.  I expect my final box will be about 300mm wide and about 75mm depth.  How can I get and use the parametric data from the web page so that the finger holes will grow along with the extra inside wall I have added?

florianfesti, 2020-07-22:
You need to turn those calls into lambdas to make sure they are actually at the right place. This also makes the placement easier as each call is made from the start of the bottom edge of the segment in question:

```
                             callback=[
                                 lambda: self.fingerHolesAt(0, y_here, (x-2*r)/2, angle=0),
                                 lambda: self.fingerHolesAt(0, y_here, y-2*r, angle=0),
                                 lambda: self.fingerHolesAt(0, y_here, x-2*r, angle=0),
                                 lambda: self.fingerHolesAt(0, y_here, y-2*r, angle=0),
                                 lambda: self.fingerHolesAt(0, y_here, (x-2*r)/2, angle=0),

```

st2000, 2020-07-22:
That works!  Excellent.  Thanks @florianfesti .  I will have to experiment a bit to understand what is going on.  So I can make my own adjustments.  But will close this issue as the question now has a solution

Ultimately I would like to alter the Python program to create the perfect box for my (Word Clock) electronics project.  Like add new openings to the top and bottom.  Providing I finish and there is any interest I can post the code here or fork the project with the altered roundedbox.py script.

-thanks

closed 2020-07-22 by st2000

florianfesti, 2020-07-22:
A lambda is basically a function. The trick is that the lambdas are called by the part (surroundingWall()) itself at the right location. Without lambdas those functions are called beforehand and only their results (None) are passed in (Basically doing the same as just calling them before the surroundingWall() call.

I am happy to add this as a separate generator as soon as it is finished.