florianfesti/boxes #230: TrayType With Sliding Lid Help

AllanKustom, 2020-07-19:
Hello. 
I'm trying to make a TrayType box but I want to create it with a sliding lid. 
https://festi.info/boxes.py/TypeTray?language=en
I can make it the size I want with the internal squares and box size as I want. 
What I want through is a nice sliding lid like the CardBox
https://festi.info/boxes.py/CardBox?language=en
How do I go about doing this as I cant seem to figure out how to do this in the TrayType setup. 
Thanks for your time and help
Allan :)

florianfesti, 2020-07-20:
There is currently no easy way of doing that.
One way to work around this limitation would be to generate a CardBox with only one compartment and glue in a TrayInsert or a TypeTray with an "e - StraightEdge" bottom_edge. But you need to be really careful with the sizes so everything really fits together.

AllanKustom, 2020-07-20:
Thank you very much I’ll give it a go! 😀

florianfesti, 2020-08-04:
I hope this worked out for you.

I am closing this as I currently don't plan to program anything in this direction.

closed 2020-08-04 by florianfesti