florianfesti/boxes #241: inkscape-boxmaker

hellerbarde, 2020-09-09:
A couple of years ago, I decided to put a desolate inkscape extension on github because it was only available as a zip file somewhere on a website that is now not reachable anymore. see here: https://github.com/hellerbarde/inkscape-boxmaker

Since then I dicovered the boxes.py website and never really looked back. 

recently a kind soul ported the code to inkscape 1.0 so I got a notification about the repository again. I now realized that boxes is also a inkscape extension. Could you have a look at inkscape-boxmaker to see if there is anything you think could be useful to your project (mostly from a inkex perspective)? I think it would be sensible if I could in the end just close down the boxmaker repo and just provide a link to your repository. There is no need for zombie projects.

florianfesti, 2020-09-15:
Well, this is actually not that easy to answer. The relevant difference between inkscape-boxmaker and Boxes.py is that the former is basically just two files that can be copied into extensions directory while the later is a bit bigger project with modules, additional files and dependencies.
This has always been an issue with Boxes.py in general but even more so for the Inkscapeplugin. This got a bit better with the new back end that no longer requires libcairo which is close to impossible to get running on Windows -  well, at least that's what I heard. The issue is to get Boxes.py packaged in a way that it can be used installed and used easily as Inkscape plugin. This is something that I probably can do on Linux but is much harder on Windows. Me not using Windows since NT and never having used MacOS is not exactly helping.
So there is not really something from inkscape-boxmaker that actually helps here.

So that's not a great answer but I guess I need to put a bit more effort into the Inkscape plugin in Boxes.py anyway to make it work on 1.0.

hellerbarde, 2020-09-15:
I would like to help. What are the current pain points, now that libcairo is gone?

Maybe I should start with checking how difficult it is to get running on the new 1.0 version on macos, since I got such a laptop somewhere.

florianfesti, 2020-09-16:
Well, this issue is that I didn't find any good documentation on how to package Inkscape extensions at all. Basically Boxes.py is a Python project. Guess at some point I really need to bite the bullet and put it on PyPi. But for me it is not really clear how to add the Inkscape extension part in there in a way things work out of the box without users needing to move files around.
I am also not sure if the extension should be labeled as a Python extension (using the Python shipped with Inkscape) or as an executable. On an Posix system where everything has it's place on the file system this is not that complicated but I really have no idea how to make everything fit together anywhere else.

Then there's the whole issue of system vs user installation.

hellerbarde, 2020-09-16:
I would like to look into proper packaging for inkscape extensions. I'll report back with what I find out.

Am I correct in my conclusion that the only libraries necessary for extension usage are `affine` and `lxml`?

hellerbarde, 2020-09-16:
Actually, lxml is not used in the code anymore, as far as I can tell. Should I remove it from the requirements?

florianfesti, 2020-09-16:
> Actually, lxml is not used in the code anymore, as far as I can tell. Should I remove it from the requirements?

yes, please!

florianfesti, 2021-06-07:
Boxes.py is now included in https://gitea.fablabchemnitz.de/MarioVoigt/mightyscape-1.X

This offers an easy way to install Boxes.py as an Inkscape plugin by simply copying some files (actually a lot of files) into the plugins folder. Guess this solve the issue at hand. Closing.

closed 2021-06-07 by florianfesti