florianfesti/boxes #242: Rounded box adds an extra line under the side

mahboud, 2020-09-13:
This is what it looks like: 

[image]
(see the tabs at the bottom)

This line continues for about three of the fours sides and then disappears.

Here's the full file:

[RoundedBox-6.svg.zip](https://github.com/florianfesti/boxes/files/5215086/RoundedBox-6.svg.zip)

florianfesti, 2020-09-14:
Ok, this is easy to reproduce. Thanks for reporting!

There are two issues here:

1. The code trying to close paths does not check whether they are interrupted. So this line connects the end of the path at the bottom left of the part to the start of the last uninterrupted segment. This is just what SVG does when appending "Z" to the path. Proper fix is to check each segment instead of the start and end of the path.
2. The flex cuts are part of the path that is the outline. This is an issue with the FlexEdge. Having them as separate paths should allow to make the outline a closed path again.

I'll try to get this fixed soon.

mahboud, 2020-09-15:
If I select "F finger joint" instead of "f finger joint", i get a slanted line that cuts through the tabs.

[image]

mahboud, 2020-09-15:
Also, on the original box, I can't get the tabs to fit each other. Nor the connectors on the sides. Everything is slightly off, and I don't recall that from the last time (over a year ago) that I  made a rounded box.

florianfesti, 2020-09-15:
> If I select "F finger joint" instead of "f finger joint", i get a slanted line that cuts through the tabs.
Yes, this is expected. The SVG "Z" command just connects the last and the first points of the line segment to create a closed loop.

florianfesti, 2020-09-15:
Anyway. The current update should fix both issues. So now the outline of that part should be continuous and closed. Thanks again for reporting.

florianfesti, 2020-09-15:
> Also, on the original box, I can't get the tabs to fit each other. Nor the connectors on the sides. Everything is slightly off, and I don't recall that from the last time (over a year ago) that I made a rounded box.

This is kinda weird. If you can reproduce this I am interested to hear about it. A link to the generated drawing is the preferred way to report stuff like this.

> (I'm going to open another ticket for the burn fit issue i mentioned here, once I reproduce it better.)
Yes, please!

But there are chances that there is some error on your side. Either the drawing wasn't rendered with the right burn settings (smaller numbers mean more slop!) or you are using a different laser or material that requires a different settings. Boxes with finger holes can also run into trouble if the material is thicker than set.
There is the BurnTest generator in "Parts and Samples" in case you want to test different settings for `burn`.

commit e35447e5 (2020-09-15): Fix FlexEdge to use separate path for the flex cut

to allow continous outlines for parts.

Resolves: #242