florianfesti/boxes #245: Working on variation of roundedbox.py: Increasing number of side panels from 1 to 2 unexpectedly eliminates internal shelf.

st2000, 2020-09-28:
Hi,

This is a continuation (related to issue 229) of me trying to make a variation of roundedbox.py to create a box suitable for a Word Clock.

After adding internal layers to the roundedbox.py design I realized I could not cut the side of the box as it was too long.  I incremented the side panel counter from 1 to 2 which created 2 shorter side panels.  However, the internal shelf disappeared from the generated cut pattern.

```
#!/usr/bin/env python3
# Copyright (C) 2013-2014 Florian Festi
#
#   This program is free software: you can redistribute it and/or modify
#   it under the terms of the GNU General Public License as published by
#   the Free Software Foundation, either version 3 of the License, or
#   (at your option) any later version.
#
#   This program is distributed in the hope that it will be useful,
#   but WITHOUT ANY WARRANTY; without even the implied warranty of
#   MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
#   GNU General Public License for more details.
#
#   You should have received a copy of the GNU General Public License
#   along with this program.  If not, see <http://www.gnu.org/licenses/>.

from boxes import *

class RoundedBox(Boxes):
    """Box with vertical edges rounded"""

    ui_group = "FlexBox"

    def __init__(self):
        Boxes.__init__(self)
        self.addSettingsArgs(edges.FingerJointSettings)
        self.buildArgParser("x", "y", "h", "outside")
        self.argparser.add_argument(
            "--radius", action="store", type=float, default=15,
            help="Radius of the corners in mm")
        self.argparser.add_argument(
            "--wallpieces", action="store", type=int, default=1,
            choices=[1, 2, 3, 4], help="# pieces of outer wall")
        self.argparser.add_argument(
            "--edge_style", action="store",
            type=ArgparseEdgeType("fFh"), choices=list("fFh"),
            default="f",
            help="edge type for top and bottom edges")
        self.argparser.add_argument(
            "--top",  action="store", type=str, default="none",
            choices=["closed", "hole", "lid",],
            help="style of the top and lid")

    def hole(self):
        t = self.thickness
        x, y, r = self.x, self.y, self.radius

        dr = 2*t
        if self.edge_style == "h":
            dr = t

        if r > dr:
            r -= dr
        else:
            x += dr - 2*r
            y += dr - 2*r
            self.moveTo(dr-r, 0)
            r = 0

        lx = x - 2*r - 2*dr
        ly = y - 2*r - 2*dr

        self.moveTo(0, dr)
        for l in (lx, ly, lx, ly):
            self.edge(l);
            self.corner(90, r)

    def render(self):

        x, y, h, r = self.x, self.y, self.h, self.radius

        if self.outside:
            self.x = x = self.adjustSize(x)
            self.y = y = self.adjustSize(y)
            self.h = h = self.adjustSize(h)

        r = self.radius = min(r, y / 2.0)

        t = self.thickness

        es = self.edge_style

        corner_holes = True
        if self.edge_style == "f":
            pe = "F"
            ec = False
        elif self.edge_style == "F":
            pe = "f"
            ec = False
        else: # "h"
            pe = "f"
            corner_holes = True
            ec = True

        with self.saved_context():
            self.roundedPlate(x, y, r, es, wallpieces=self.wallpieces,
                              extend_corners=ec, move="right",
                              callback=[self.rectangularHole(50,y*.5,10,y*.8)])
            self.roundedPlate(x, y, r, es, wallpieces=self.wallpieces,
                              extend_corners=ec, move="right",
                              callback=[self.rectangularHole(50,y*.5,10,y*.8)])
            self.roundedPlate(x, y, r, es, wallpieces=self.wallpieces,
                              extend_corners=ec, move="right")
            self.roundedPlate(x, y, r, es, wallpieces=self.wallpieces,
                              extend_corners=ec, move="right",
                              callback=[self.hole] if self.top != "closed" else None)
            if self.top == "lid":
                r_extra = self.edges[self.edge_style].spacing()
                self.roundedPlate(x+2*r_extra,
                                  y+2*r_extra,
                                  r+r_extra,
                                  "e", wallpieces=self.wallpieces,
                                  extend_corners=False, move="right")

        self.roundedPlate(x, y, r, es, wallpieces=self.wallpieces, move="up only")

        y_here = h/2
        if self.wallpieces == 1:
            self.surroundingWall(x, y, r, h, pe, pe, pieces=self.wallpieces,
                                 callback=[
                                     lambda: self.fingerHolesAt(0, y_here, (x-2*r)/2, angle=0),
                                     lambda: self.fingerHolesAt(0, y_here, y-2*r, angle=0),
                                     lambda: self.fingerHolesAt(0, y_here, x-2*r, angle=0),
                                     lambda: self.fingerHolesAt(0, y_here, y-2*r, angle=0),
                                     lambda: self.fingerHolesAt(0, y_here, (x-2*r)/2, angle=0),
                                 ])
        y_here = h*3/4
        if self.wallpieces == 1:
            self.surroundingWall(x, y, r, h, pe, pe, pieces=self.wallpieces,
                                 callback=[
                                     lambda: self.fingerHolesAt(0, y_here, (x-2*r)/2, angle=0),
                                     lambda: self.fingerHolesAt(0, y_here, y-2*r, angle=0),
                                     lambda: self.fingerHolesAt(0, y_here, x-2*r, angle=0),
                                     lambda: self.fingerHolesAt(0, y_here, y-2*r, angle=0),
                                     lambda: self.fingerHolesAt(0, y_here, (x-2*r)/2, angle=0),
                                 ])
        if self.wallpieces == 2:
            self.surroundingWall(x, y, r, h, pe, pe, pieces=self.wallpieces,
                                 callback=[
                                     lambda: self.fingerHolesAt(0, y_here, (x-2*r)/2, angle=0),
                                     lambda: self.fingerHolesAt(0, y_here, y-2*r, angle=0),
                                     lambda: self.fingerHolesAt(0, y_here, (x-2*r)/2, angle=0),
                                     lambda: self.fingerHolesAt(0, y_here, (x-2*r)/2, angle=0),
                                     lambda: self.fingerHolesAt(0, y_here, y-2*r, angle=0),
                                     lambda: self.fingerHolesAt(0, y_here, (x-2*r)/2, angle=0),
                                 ])
```

florianfesti, 2020-09-28:
Well, this is kind of expected - although not that convenient for you.
The issue here is that the callbacks are per side. But the number of sides varies depending on the `pieces` param as the sides get split into two. This shifts the numbers of the callbacks around.
Basically you need one set of callbacks for each value of `pieces`.

You have tried to do that above. But it looks like you are just painting the same part over the same place. That's not ideal. Try using a actual function as a callback that draws the fingerholes for all heights at once.
Instead of a list you can also pass a function directly as `callback` param. It then gets the number of the callback position as an itn param. This might be more convenient in this case as you can decide what to do depending one the value of `wallpieces`.

bwente, 2020-10-08:
@st2000 I would be interested to see what you come up with. I used the rounded box as temple for a word clock enclosure too. 

https://www.hackster.io/bwente/word-clock-with-just-two-components-073834

scroll to the image at the bottom for the rounded box version

florianfesti, 2021-05-23:
I don't think there is going to be any change in how the callback work for the surrounding wall. While the number could be made more consistent most callback will still need to take precautions on the number of pieces. Closing as WONTFIX.

closed 2021-05-23 by florianfesti