florianfesti/boxes #251: Option to remove "burn correction" for milling

ottobot42, 2020-10-06:
I'd like to produce boxes, otto-bots and the like by using an endmill. 

When setting the bur correction parameter to 0.0 I get a very loose fit of the parts.
With positive values  fro burn correction as suggested for tighter fit there are additional paths generated at corners which are not needed for milling (in fact they are a pain when generating the gcode).

Is it feasible to introduce an option to disable the burn correction feature for milling?

florianfesti, 2020-10-06:
Well, technically those additional paths are actually needed to make sure the corners are milled out. 

You can set the burn param to 0.0 and do the kerf correction with another tool. Most CAM software should be able to do that. Not that I have any experience with that.

There is some outdated documentation on this topic at https://florianfesti.github.io/boxes/html/api_burn.html

The new back end now optimizes the path for internal  corners in https://github.com/florianfesti/boxes/blob/master/boxes/drawing.py#L188 replacing back facing arcs by Bezier loops. (Yes, the code is not properly commented and difficult to understand). It would be easy to replace them with something different. What path would you suggest instead?

ottobot42, 2020-10-06:
I use Estlcam an Fusion360 to generate the gcode. They do have an option to overcut in corners.

When I set the burn param to 0.0 I get a plai path but the resulting pieces fit together very loosely. 
Measuring the fingers of a 3mm Plywood part they are 2.8mm wide and the gaps are 3.2mm wide instead of 3.0mm and 3.0mm as I would prefer. 
I did some other projects with fingerjoints in Fusion360 and those fingers were a snug fit. Thus I suspected that you apply a correction based on the burn parameter. My assumptions might be wrong though.

florianfesti, 2020-10-06:
The burn parameter is half the width of cut aka cutter radius aka kerf. In Boxes.py the width of the lines show the what is actually cut away (assuming the burn value is correct).

Unless you can do cuts with zero width you need to compensate for the width of the cut somehow. While I am not familiar with Estlcam I am pretty sure you can tell it the diameter of the milling bit somewhere. If you set burn to 0.0 and set your milling bit size correctly everything should just work. Alternatively you can set the burn value to the radius of you bit and tell your CAM software to stay on the lines.

ottobot42, 2020-10-07:
OK. I think I get the idea: setting burn to 0.0 should produce the exact geometry. So with a material thickness of 3.0mm all fingers and gaps should be of 3.0mm width and height respectively.

I used a 1.0mm drill bit and I rechecked my settings in Estlcam were correct. CAM software is generally taking care to offset the path of the drill bit by half of its diameter. 
 

I did some further tests importing the DXF and SVG into Fusion360 and CorelDraw X6:

In Fusion360 dimensions of the fingers show that SVG output is correct ( horizontal and vertical fingers and gaps are 3.0mm wide). 
[image]

DXF (in Fusion 360) is imported as single lines and vertical fingers and gaps are 3.0mm but horizontal fingers are 2.7mm and horizontal gaps are 3.3mm wide. The height of fingers however is 3.0mm for both horizontal and vertical fingers.
[image]

- When I import the DXF into CorelDraw X6, I get roughly the same measurements as in Fusion360.
[image]

For me it is a bit harder to get exact dimensions there. This might be due to the fact that the individual lines seem to be kind of double layered here. The object properties of a line show that it has been imported as an open curve comprised of 6 nodes and 2 lines.
[image]

My usual workflow is to rearrange shapes/objects in CorelDraw before generating drill bit paths and gcode with Estlcam. 
Importing/modifying/exporting DXF with CorelDraw each line is shown as a very slender rectangle in Estlcam. This prevents Estlcam from recognising the outlines as they are not a closed sequence of single lines.

At the end of the day the DXF export from Boxes.py might not be suited as input for milling, unfortunately. So I might have to switch to SVG instead.

Thank you Florian for your time and consideration.

Blancanegra, 2020-10-07:
Hi, I'm using Corel Draw and SVG format works like a charm. Use 0.0 burn setting for an exact geometry size.
In previous builds, SVG paths were open and I used join curves tool in Corel Draw to close the paths.
In lastest build of boxes.py SVG paths are closed when imported in Corel Draw but in your case I think you must combine the paths of pieces with inner holes, then Estlcam would generate the proper inner and outter tool compensation.

I don't use Estlcam but VCarve, sure there is a toolpath option to mill over the path line. In that case you can use the burn setting  in boxes.py according to your tool radius, simplifying your workflow.

The only thing here, I'm not sure if Corel Draw 6 is managing the SVG files as latest versions would do.

ottobot42, 2020-10-08:
@Blancanegra: Thanks for your input. 
When I use SVG I need to rescale. That's why I prefer DXF.
But maybe I'm missing something here.

florianfesti, 2020-10-19:
I hope you found a work flow that is sufficient for you. I currently don't see what else can be done here. So I am closing the ticket.

closed 2020-10-19 by florianfesti