florianfesti/boxes #256: unrecognized arguments on MakitaPowerSupply

PaoloElMarinelo, 2020-11-15:
Hi everyone,
Trying to generate https://festi.info/boxes.py/MakitaPowerSupply?FingerJoint_angle=90.0&FingerJoint_style=rectangular&FingerJoint_surroundingspaces=2.0&FingerJoint_edge_width=1.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&bananasocketdiameter=8.0&flipswitchdiameter=6.3&thickness=3.0&format=svg&tabs=0.0&debug=0&reference=100&burn=0.1&render=1
Feedback is : **unrecognized arguments: --bananasocketdiameter=8.0 --flipswitchdiameter=6.3**
Hoping to be able to be useful to you through this request
Thanks 
Paolo

florianfesti, 2020-11-15:
I actually had fixed that already just a few days ago:
https://github.com/florianfesti/boxes/commit/09b4de82def027e663ae0d9c864a2fba920f9dde

But I didn't update the web instance right away. So it should work now.

I also didn't yet add the picture I finally took.

closed 2020-11-15 by florianfesti

florianfesti, 2020-11-15:
There it is.

florianfesti, 2020-11-15:
Anyway, thanks for reporting!