florianfesti/boxes #259: TypeTray finger joints not sizing correctly

h0rn37, 2020-11-18:
I've used the TypeTray before with zero issues but today I've attempted it 4 times, each resulting with the same problem.

1. Created TypeTray as usual and attempted to construct. Finger joints wouldn't connect (I forgot to change material thickness, this was obviously my fault)
2. Adjusted material thickness and tried again. Finger joints still wouldn't fit together right
3. Adjusted burn from .1 to .5. In the past, this allows my boxes to fit together with slight use of rubber mallet. Finger joints still wouldn't fit together
4. Adjusted burn to 0. In the past, this allows the boxes to fit together very loosely and require wood glue. Finger joints still wouldn't fit together.

After all of these, I decided to throw it into Inkscape to see what was going on. The measurement difference between the finger joint length and the hole is 1mm. This was consistent in every version of the file I attempted. Obviously I can go in and adjust manually with the vector tool, but I figured people might want to know it's acting up so they don't expect it to work as normal.

florianfesti, 2020-11-19:
Sorry to hear you had so much trouble. Can you please provide the link from where you downloaded the file. What format did you use?

florianfesti, 2020-11-19:
This sounds awfully like https://github.com/florianfesti/boxes/issues/251#issuecomment-704766234
Unfortunately I have still not found out what's the problem there. I replaced the back end a few months ago. But the dxf is still converted from the PostScript output. May the pstoedit doesn't like the new format?

h0rn37, 2020-11-19:
> This sounds awfully like [#251 (comment)](https://github.com/florianfesti/boxes/issues/251#issuecomment-704766234)
> Unfortunately I have still not found out what's the problem there. I replaced the back end a few months ago. But the dxf is still converted from the PostScript output. May the pstoedit doesn't like the new format?

Sorry, I don't have the windows open anymore so I can't link to it. I tried to attach the files, but github is giving me a hard time about it. I downloaded them as SVGs to answer your previous question.

florianfesti, 2020-11-19:
Yeah, you need to put it in a zip file to be able to upload it. The SVG also has the link in a comment. May be just open the file with an text editor and past the comment here.

florianfesti, 2021-05-23:
No news in a while. Closing for now. Feel free to re-open if you have additional information.

closed 2021-05-23 by florianfesti