florianfesti/boxes #260: bug in gears generator

nialabert, 2020-11-25:
Seems to have a bug in the gears generator.
I'm not sure but I think it's a confusion and/or mix between module and pitch.

If you stays with the default setting:
Z = 12 
m =  5

You should have 
Dp  = 60
De = 70

But you can find displayed on the svg:
Pitch radius 1: 9.5mm
Outer diameter 1: 21.6mm

Almost an error of π

florianfesti, 2020-12-24:
After looking at this I am wondering if one of us confuses diameter an circumference. Surprisingly there is exactly a factor π between them.

nialabert, 2021-01-10:
mmmhhh why the circumference appear in this discussion? 

modulus of 5, number of teeth of 12 should have a outside diameter of 70mm not 21.6mm.
The gear generated is clearly not with the right size.

Sometime the pitch defined as π·m appear, but it's confusing. You can see that on the French page of wikipedia about gears, but not on the English one.

florianfesti, 2021-01-10:
OK, looks like the module I derived the code from uses this wrongly. Need to have a look where else this needs to be fixed.

nialabert, 2021-01-21:
Have a look when the pitch is used. It shouldn't. Thanks

florianfesti, 2021-06-26:
OK; this took way too long for a one line fix. Last time I got lost in the code somehow.

Anyway, thanks for reporting!

commit 62389032 (2021-06-26): Gears: Use proper metric module

with diameter / #teeth instead of circumference / #teeth.

Thanks to nialabert (https://github.com/nialabert) for pointing this
out!

Resolves: #260
closed 2021-06-26 by florianfesti