florianfesti/boxes #262: Cannot figure out how to instantiate edges in UniversalBox (python API)

kdoroschak, 2020-11-30:
Hi! I am trying to use UniversalBox through the python API instead of through the web form but am having issues.

TL;DR I first instantiate the box, parse arguments, and then call `box.render()`. I get an `AttributeError: 'UniversalBox has no attribute 'edges'.`

**Short code example to produce the error**:

```
from boxes import generators, edges

generators.getAllBoxGenerators()

box = generators.universalbox.UniversalBox()

top_edge = "L"
bottom_edge = "F"
x = 100.0
y = 100.0
h = 100.0
outside = True
vertical_edges = "finger joints"
lid = "default (none)"

burn = 0.1
thickness = 3.0

args = f"--top_edge={top_edge} --bottom_edge={bottom_edge} --x={x} --y={y} --h={h} " +\
       f"--outside={outside} --vertical_edges={vertical_edges} --lid={lid} " + \
       f"--burn={burn} --thickness={thickness} --debug=True"
args = [f"--{x.strip()}" for x in args.split("--") if len(x) > 0]
print(args)
box.parseArgs(args)
box.render()
```

**Args look like this**:

`['--top_edge=L', '--bottom_edge=F', '--x=100.0', '--y=100.0', '--h=100.0', '--outside=True', '--vertical_edges=finger joints', '--lid=default (none)', '--burn=0.1', '--thickness=3.0', '--debug=True']`

I also tried including the args for edges & the lid. There was no change in behavior so I left it out for brevity.

**Here's the full error**:

```---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-2-6df54a0903e7> in <module>
----> 1 box.render()

~/Documents/side_projects/board_game_boxes/boxes/boxes/generators/universalbox.py in render(self)
     63         t = self.thickness
     64 
---> 65         t1, t2, t3, t4 = self.topEdges(self.top_edge)
     66         b = self.edges.get(self.bottom_edge, self.edges["F"])
     67 

~/Documents/side_projects/board_game_boxes/boxes/boxes/lids.py in topEdges(self, top_edge)
    103 
    104     def topEdges(self, top_edge):
--> 105         t1 = t2 = t3 = t4 = self.edges.get(top_edge, self.edges["e"])
    106 
    107         if t1.char == "i":

AttributeError: 'UniversalBox' object has no attribute 'edges'
```

I searched through the documentation & source, trying to find an example or some configuration step that I missed (maybe similar to the need to call `generators.getAllBoxGenerators()`?). Also tried manually instantiating that variable but since dictionaries are nice little black boxes (I think it's supposed to be a dictionary anyways), reverse engineering isn't directly feasible.

Could you please help me figure out what's missing here? Thanks so much.

florianfesti, 2020-12-01:
Well, there already is `scripts/boxes` that offers a command line interface. But if that's not what you need you can have a look inside and copy the code from `run_generator`: https://github.com/florianfesti/boxes/blob/master/scripts/boxes#L64

kdoroschak, 2020-12-01:
Thanks! Looks like I was missing `box.open()`.

closed 2020-12-01 by kdoroschak