florianfesti/boxes #263: language change box is empty

hznupeter, 2020-12-10:
I have install boxes on my computer,but I can't change the language at bottom.
how to change the code or setting?
[image]

florianfesti, 2020-12-10:
This is probably cause by boxes.py not finding the language files:

```
locale/en/LC_MESSAGES/boxes.py.mo
locale/fr/LC_MESSAGES/boxes.py.mo
locale/zh_CN/LC_MESSAGES/boxes.py.mo
```
How did you install Boxes.py and under which operating system? It's quite possible some variants still miss places those files. May be you can find them some where on disk and I can just add that to the search path.

hznupeter, 2020-12-11:
i have fix this problem by install on another computer.
i want to help you translate to chinese. but i cant edit on https://translate.zanata.org

mahmyd, 2021-01-19:
> i have fix this problem by install on another computer.
> i want to help you translate to chinese. but i cant edit on https://translate.zanata.org

You can. Just connect to chinese team before. 
Login to zanata, go to Setting - Languages, click on Join a language team, find language what you need, open it and click on Join Team button 
[image]

florianfesti, 2021-05-30:
Looks like the original issue is solved. Closing.

closed 2021-05-30 by florianfesti