florianfesti/boxes #264: Upgrading to Fedora 32 breaks the boxes extensions

jgforbes, 2020-12-11:
After upgrading OS the Fedora 32 from 31, running a boxes extension gives Inkscape  error box with traceback:
Traceback (most recent call last):
  File "/usr/bin/boxes", line 4, in <module>
    __import__('pkg_resources').run_script('boxes==0.1', 'boxes')
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3252, in <module>
    def _initialize_master_working_set():
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3235, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3264, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 583, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 900, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 786, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'boxes==0.1' distribution was not found and is required by the application

I think that it is related to the upgrade to python3.8 but I have not found how to fix the problems.
Any ideas?

jgforbes, 2020-12-11:
I also installed the latest version with no effect.

jeffforbes, 2020-12-15:
Fixed by putting the latest boxes runnable in /usr/local/bin and the boxes directory in /usr/local.

florianfesti, 2020-12-17:
Hmm, how exactly did you install Boxes.py and where did the files and up. Unfortunately things need to go different places for different installation types and operating systems.

matkoniecz, 2021-03-02:
ping @jgforbes https://github.com/florianfesti/boxes/issues/264#issuecomment-747435464

florianfesti, 2022-01-28:
Guess it works now. Closing.

closed 2022-01-28 by florianfesti