florianfesti/boxes #266: Castle render

mattyzell, 2020-12-14:
I just signed up for GitHub to try to contribute an image (well, a rendering actually).
[image]

florianfesti, 2020-12-19:
Thanks for all the effort. This is a very rudimentary thing I did as a table decoration for a medieval style wedding. I have ever since dreamed of this turning into a proper castle but I never real had the time for that and I really can't make any promises right now.

closed 2020-12-19 by florianfesti

mattyzell, 2020-12-19:
I'm not sure if that image was helpful or not.  I'm just starting to tinker
with laser cutting and the related software.
Cheers!

commit 09bbfcac (2020-12-19): Castle: Add rendering and quick comment

Thanks to mattyzell (https://github.com/mattyzell) for the randering

Resolves: #266