florianfesti/boxes #267: StorageRack mixes up finger joint dimensions

impellerturn, 2020-12-29:
Hi,
first of all, thanks for your great work.

When assembling my version of the StorageRack (see https://github.com/florianfesti/boxes/issues/140#issuecomment-752043778) I noticed that the top and bottom part do not fit to the rear wall. Had to cut it manually with a cutter-knife.
When analyzing the generated SVG I noticed, that the dimensions (8,2mm and 7,8mm) were used wrong for the finger joints.
I´ve generated the SVG via https://www.festi.info/boxes.py/StorageRack?language=en on 21st December 2020.

Remark: all the other joints did fit very nicely.

Here is the SVG: 
[StorageRack.zip](https://github.com/florianfesti/boxes/files/5750575/StorageRack.zip)

Here the screenshots from inkscape where the mixed up dimensions are visible:
Top/bottom part:
- [image]
- [image]
Rear wall part:
- [image]
- [image]

Happy new year 2021 to all, best regards

florianfesti, 2020-12-29:
Hmmm, I am not quite getting what the problem is here. The difference of these measurements is due to the burn correction (which is set to 0.1mm). So holes and recesses are expected to be 0.1mm smaller on each side while fingers are expected to be 0.1mm wider on each side. If the fit is too tight you may want to *decrease* the burn parameter. May be 0.08 or 0.06 work better for you. You may use the BurnTest generator in "Parts and Samples" to figure out the best value.

I am a bit puzzled that all other joints are fine in your opinion as the look very similar. May be you also have a problem with the material thickness not quite being on the nominal value.

impellerturn, 2020-12-29:
Yes seems legit. Dimensions are as all the other finger joints.
The  fit was good for all the other joints.
Maybe some axis calibration issue with the machine. I´ll test that.
For now I would close this ticket.

closed 2020-12-29 by impellerturn

florianfesti, 2020-12-29:
It is possible that the laser beam is slightly tilted in one axis or the beam is not round and though needs different corrections in different corrections. We had two or three evening long calibration sessions until our machine was aligned perfectly.

impellerturn, 2020-12-29:
I assume that the linear movement of my machine in X is not exactly as in Y. Some people in the corresponding forum suspect that the screw-drives of the linear movement actuators have different slope/inclination due to manufactoring errors.
I want to engrave a square and measure the outcome to proof that. Maybe i´ll need to correct the steps/mm settings of my machine on one axis.