florianfesti/boxes #270: Make fingers longer

rokelvisar, 2021-01-10:
Make input param that would make fingers longer by input value - that would be perfect setting for those that want to sand down burn marks at laser cut boxes.

Azzman77, 2021-11-28:
This woul be great!

StevenHirst, 2022-01-13:
Given that most parameters are in terms of material thickness, you could set the 'finger length' to say 1.1 (which would be 4.4mm in your case) if it were implemented in this way.

StevenHirst, 2022-01-14:
I've thought a bit more about this, and:

I use material that is 3.2mm.
Set material thickness to 4mm
In finger join settings set 'play' to -0.2

This seems to work well for boxes, but not so well for trays (need to close up the gap from 4mm to 3.2mm).

Photo to follow...

florianfesti, 2022-01-16:
OK, I finally added another parameter to the Finger Joint settings. Last time I tried that I looked at the wrong place and was intimidated on what it would take to make this really work. Turns out there is a much better place.

Thanks to everyone pushing for this. Please test the new `extra_length` setting!

commit a364f603 (2022-01-16): Add extra_length to Finger Joint Settings

to allow extending the finger joints so burn marks can be ground off.

Thanks to all who have been pushing for this feature!

Resolves: #270
closed 2022-01-16 by florianfesti