florianfesti/boxes #274: DividerTray wrong box length

Feredyr, 2021-01-27:
Hello,
Thank you for this amazing work !
I try to use the DividerTray but I have an issue when I setup my box.
The box length did not fit what I setup:
[image]

The result
[image]

Could you please help me ?

Thanks!

gcollic, 2021-01-27:
Hello !

When the "exterior" parameter is enabled, you are telling this box what should be the total length.

But when the "exterior" parameter is disabled (like in your example), you are telling this box what should be the length of the content for each slot. The last slot ends at the right place, but requires some extra space above it, and I think it explains why it's not the size you wanted:
[image]

Most of the time, you both want precise information about inner size (how much stuff each slot needs to store) and outer size (how much space I have for the entire box). I added some information in debug mode to help find the right parameters to have both the right inner and outer sizes.

Can you give the "debug" checkbox option a try and tell me if it helps ?

Don't hesitate to tell us how we could improve this configuration for clarity. "Outside" and "debug" are used by other boxes, the  text is not as helpful as it could be for this particular box, I might need to create a new parameter of the same name but a different description. Or I might just add some text to the description (it goes below the form, with the picture).

Thank you !

gcollic, 2021-01-27:
It looks like there is still an error after all. The second slot is bigger than asked and than the first. I should look into that !

Feredyr, 2021-01-27:
Thank you ! !
Enabling the exterior parameter fix almost the issues. 

 From my point of view, the text could be clarify a little bit.
if I may help:
This kind of box is mostly done for board game insert (Like in my case). So the X / Y / Z exterior parameters are mandatory to really fit the cardboard box. Then we divided the by number of section ( Sx). 
Where there can be mistake is with the Z (H) parameter.  Without the exterior checkbox, the H is the card length (in the tray). Maybe you could remove the slot angle parameter and ask for card length (as this is just math)
One other improvement than can be done, is to make start the Slot against one wall. That’s avoid space lose.
Again thank you for this amazing work

closed 2021-01-28 by Feredyr

gcollic, 2021-01-28:
> start the Slot against one wall

@Feredyr you can use a 0 sized slot first and it should work properly :-) For example "0:270/2", or "0:270/2:0".

gcollic, 2021-01-28:
> This kind of box is mostly done for board game insert (Like in my case).

I created it for this exact  purpose :-)

> So the X / Y / Z exterior parameters are mandatory to really fit the cardboard box. Then we divided the by number of section ( Sx).

Well, it depends, in fact. You probably need the inner boxes to be smaller than the game box, but not always the exact same size if you need to also fit other game components in other boxes. Sometimes you know the size of the decks of cards you need to store for this particular game, then it generates a box which fits all of those cards, and you measure how much space you are left with to create other boxes for the other components in the remaining space :-)

And sometimes you create a whole new game box which doesn't need to fit into the old ones (collectible card games, etc).

And even some time, there are so much components with game extensions that you need a bigger game insert than the original game box (the lid will not totally close, but it's done this way in a lot of cases, even for professional custom game insert).

All the variables/parameters depend on each other (outer x, outer y, outer z, deck height, card width, card height, angle, + slots). You need any 4 information (+ slots) to compute the others, and depending on the project I don't think it will always be the same ones that are fixed. I thought it was simpler to only propose the same 2 references as the other boxes : outside or inside, but sadly that means you need some conversion (by math, or by experimenting and observing the numbers in the "debug" part). If we let more parameters, I'm not sure how to explain only 4 of them are needed in any combination ? Do you have some idea how we could improve your use case configuration without degrading the other ?

Did you try the "debug" option ? I find it really helpfull for the same purpose you are describing :-)

gcollic, 2021-01-28:
> Enabling the exterior parameter fix almost the issues.

@Feredyr I found the problem for the remaining length difference, a fix has been proposed :-)