florianfesti/boxes #277: Typetray  Lid proposal

Feredyr, 2021-01-29:
Hello,
Again Thank you for this amazing work. Since I discover your generator, I can't stop using it.
My main purpose with it, is to create boardgame Insert.
The typetray (as the dividertray) are amazing for that.
I want to propose you one other lid format for the typetray, to save even more space in the  cardboard box.
A flat lid with fingerjoint (but only on the outside wall)  and a notch to easy remove the lid
like this one:
[image]
[image]

what do you think?

Feredyr, 2021-01-30:
Ok, so after many tries, I managed to get almost what I want. 
So now I have a new question...
Could we set different fingerjoint settings for the top and the bottom of the box and have a setting to tranforme the grip into a notch ? (or place it at the hieght we want)

ccrome, 2023-01-06:
Ooh, I like this idea for the [gridfinity generator I just made](https://github.com/florianfesti/boxes/issues/489) !

w1ml33na3rts, 2023-04-01:
Is this durable enough to remove on a regular basis? Sounds like a great idea. Right now the slide on lid doesn't generate properly so this would be a nice alternative to try. I'm just starting to experiment for the same purpose as you.

florianfesti, 2023-04-07:
Added Lids to both TypeTray and the Gridfinity generators (and all other sort of boxes and trays). The style requested here is still not available as it'd need a new edge type (and we already have enough to make things complicated). But I am open to add more lid styles.

PhoenyxAshe, 2023-05-02:
It wouldn't be a perfect solution, but could you possibly do the "on top" lid with a "reversed stackable" (ie., tabs facing down from the lid, indent on the top of the box) as an option?  It's not a complete match to what's in the OP's pictures, but it would be close.