florianfesti/boxes #278: Feature Request: TrayLayout Options, Dispenser Generator

havocmayhem, 2021-02-12:
Would it be possible to make a few enhancements to the 'TrayLayout' Generator

- Add in an option for different kinds of lids and bottoms, similar to 'TypeTray'
- Allow for an angled bottom, with the angle specified by the user. Massive bonus if you can somehow have a variable angle for each sector of the tray. 
- Allow for variable bottom layer heights for each segment.

From a UI perspective, the angles or layer heights can perhaps be input in the following way
[image]

The goal is to be able to build things like this:
[image]

Separately, could you maybe work on a 'Tile Dispenser' Generator?
This would make something like this:
[image]
Customisable parameters here would be the basic dimensions, tile thickness, width of the central gap, etc. Ideally, the bottom plate would be angled slightly upwards towards the front, to prevent the tiles slipping out unless they are pulled manually.
Bonus points if the generator can allow for multiple such dispensers placed side by side, in a single unit.

havocmayhem, 2021-02-12:
Here's another example of a tray with sloping bottoms. This makes it much easier to remove small tokens.
[image]

florianfesti, 2022-01-28:
The TrayLayout works differently than TypeTray. As all walls and floors are optional every wall can become an outer wall. So some features are not possible there. This includes different top and bottom styles. Adding those would end up in trays that cannot be assembled.

Having angled floors to allow picking up small parts is interesting but I am not sure if it can be done with TrayLayouts. May be if all floors are angled in the same direction.

florianfesti, 2022-01-28:
For the Dispenser generator: See https://festi.info/boxes.py/Dispenser?language=en

Be aware that by having requested this new generator you are obligated to do a test cut and to provide a good example picture.

havocmayhem, 2022-01-31:
Wow, thanks!

I've since moved on to using Onshape to design things from scratch, but I
hope this will be useful for other people starting out.

florianfesti, 2023-01-11:
See new SmallPartsTray generator (still in Unstable) and #497

florianfesti, 2023-01-17:
OK, there now is [SmallPartsTray](https://festi.info/boxes.py/SmallPartsTray?language=en) and [SmallPartsTray2](https://festi.info/boxes.py/SmallPartsTray2?language=en). I still have some ideas for TrayLayout but this should be enough for this ticket. Closing.

closed 2023-01-17 by florianfesti