florianfesti/boxes #279: Enhancements for DisplayShelf for creating draw pile holders/dispensers

Ungoose, 2021-02-13:
I've been using boxes.py to create board game organizers for some time now, and it's amazing. Recently, One thing I've been doing a lot is a card holder. There is a "notesholder" template that can help with it but it lacks one important feature - setting the bottom at an angle so that cards would slide back slightly. 
DisplayShelf template is particularly good for this as it allows to specify the angle of the shelf and the height of the compartment. After setting the number of shelves to one and fiddling with the result, I was able to do this:
[image]
[image]
There are several additional parameters that would make it even handier:
1) Allow dividing the shelf into sections with configurable sizes like in TypeTray (sections left to right in mm.)
2) Add the ability to create a finger hole in the middle of each section and set its size and shape (e.g., round, trapezoid\half-hex, etc.)
3) Add the ability to specify the shape of walls on the open side (rounded, angled)
4) Add the opening like in Notesholder template (currently in the back of the shelves, useful for discard piles)

If that is too much of a departure, maybe creating a separate template can be a better idea.

A very similar enhancement for TrayLauout is mentioned in #278

ZenithDK, 2021-02-13:
Can you write the parameters you have used for this? If they can be made stackable these would be great!
Just need to engrave them as topping for each game and I'd be happy 😊

Ungoose, 2021-02-14:
@ZenithDK Here's how you do it:
In DisplayShelf, choose
inner width in mm - the width of your cards / desired width of the holder
inner depth in mm - the height of your cards / desired length of the holder
inner height in mm - the desired outside height of the holder
number of shelves - 1
height of front walls - the height of the card stack / desired height of the wall in the back (keep in mind this is the outside height and subtract the thickness of the material unless you want to end up with the ugly protrusion like on the image above)
angle of floors - the desired angle of the bottom (5 in the example)
The cutout in the front has to be done manually (hence enhancement 3).

Ungoose, 2021-02-14:
Now that I think of it, most of these enhancements won't work for display shelves, so it's probably a better idea to make it a separate generator with the same set of parameters (except for the number of shelves, which will always be 1)

What would be even more awesome is to add the number of shelves but arrange them in a staircase-like fashion (for the lack of a better term) rather than one above another. This way we can generate holders like those shown in #278 and set variable bottom angles for each shelf.

florianfesti, 2022-07-30:
Note that technically you are required to provide pictures for the new generator. But as this request has taken a while this is now more or less optional.

commit 05fc5aab (2022-07-30): New Generator: CardHolder

Resolves: #279
closed 2022-07-30 by florianfesti