florianfesti/boxes #281: [feature request] add the generator settings in the svg and allow to import back in the UI

philippejadin, 2021-02-22:
Hello, great tool we have been using for years now, thanks!

An idea I had to help refinement of projects :

- add all the generator settings as metadata in the generated svg file (size, type of box, etc...)
- allow to import the settings from an existing svg file in the UI

florianfesti, 2021-02-22:
Actually the settings are already added to the SVG as an XML comment. You can have a look by just looking at the page source or opening the file in an text editor. You can also just copy the URL from there to get back to the UI.

But this information should probably also go into an Inkscape style meta data section. Any suggestions for the format?

philippejadin, 2021-02-22:
Sorry I wasn't aware of that. This is great news!

I have no suggestion for inkscape style metadata.

The last part of my idea could still apply : allow to drop an existing SVG file and reload the interface with the existing settings. But this is maybe a niche use case so feel free to close the issue.

Oh, another idea : provide a visible text link in the SVG that links back to the UI. Something like :

`<a href="https://www.festi.info/boxes.py/RackBox?FingerJoint_angle=90.0&[...]">Created with Festi box maker</a>`

florianfesti, 2021-02-22:
Oh, actually there even is RDF meta data in the file already. I just overlooked that as I was scanning for XML comments only. In Inkscape you can go on `Document properties` and then select the `metadata` tab. The data is a bit sparse but the field don't really lend themselves to be used by Boxes.py. May be som "Create by Boxes.py <link>" could be added to the description.

florianfesti, 2021-03-27:
I guess feature wise there isn't really anything new needed.But the meta data is too hard to find. May be I finally need to add an FAQ where such things could be located.

florianfesti, 2021-05-23:
OK, finally added an FAQ and mentioned the meta data there: https://florianfesti.github.io/boxes/html/faq.html#what-settings-were-used-to-generate-a-drawing

closed 2021-05-23 by florianfesti