florianfesti/boxes #283: Make TwoPiece less confusing "(unless outside selected)"

matkoniecz, 2021-02-24:
Right now on default load it has parameters such as

> inner width in mm (unless outside selected)

outside is selected by default

Maybe it would be better to have

> outside width in mm (if 'outside' is not selected then it is about the inner size)

or disable outside by default?

Sadly I was unable to find where it is defined. If you agree that it would be improvement and point in the right direction I can make a PR.

florianfesti, 2021-03-01:
Well, the box (as many others) uses some default parameters. So the description is not always the best. They can be found in `buildArgParser` https://github.com/florianfesti/boxes/blob/master/boxes/__init__.py#L359. To improve the UI one would replace the default parameter by a custom one directly using argparse.

florianfesti, 2021-04-21:
After thinking about this a bit more I think the description is good enough. The fact that `outside` is the default for this generator doesn't make the parameter really different from other in generators with `outside` off by default. Having the same description everywhere is probably less confusing. Closing.

closed 2021-04-21 by florianfesti