florianfesti/boxes #288: Why burn test generates multiple toothed rectangles?

matkoniecz, 2021-03-01:
From description:

> To get an idea cut a rectangle with known nominal size and measure the shrinkage due to the width of the laser cut.

what I understand as "cut rectangle with your laser cutter, measure real width, compare with intended width to get info about shrinkage"

But burn test generates multiple pieces with joints, in multiple variants. It looks like burn test is intended to be done by checking which pieces fit (?? is it correct ??).

But then why "compare with intended width to get info" part? Maybe it describes two possible ways to measure proper burn parameter?

florianfesti, 2021-03-02:
Yeah, the description is probably not the best. The method with measuring a rectangle is to get a first estimate for the `burn` parameter. In practice you might want to use a different value to choose the fit of the parts. For wooden parts I often chose the value 0.02 to 0.06 bigger to get a stiff press fit. For brittle material like acrylic you may want to use a slightly smaller value to get a loose fit that allows room for error and glue. The fit is something that needs to be experienced. So the generator creates finger joints with a range of settings.

matkoniecz, 2021-03-02:
> So the generator creates finger joints with a range of settings.

And I am supposed to try which ones of burned finger joints fit well, right?

So if 0.100mm fits 0.100mm then my best burn parameter is ???? and if 0.120mm side fits 0.120mm side then my burn parameter is ???? (for given laser/material/power/speed combination).

florianfesti, 2021-03-02:
Yes, you try sides with the same value and use the value you like most as `burn` parameter on the box you actually want to cut.

PR #290 (2021-03-02): try to add explanation how burn test works
fixes #288

Probably can be improved further, but I am out of ideas for now.

For images: 

- do you prefer pull request or just posting image on issue tracker?
- Do you prefer to get only images of high quality or is it OK to send also mediocre ones?
commit 21ca3d0b (2021-03-02): try to add explanation how burn test works

fixes #288
closed 2021-03-02 by florianfesti