florianfesti/boxes #289: Why burn test lines get wider as burn parater increases?

matkoniecz, 2021-03-01:
I expected shape to change if it is increased.

But primary effect that I see for increased burn parameter is that lines got wider. What is quite surprising. For start at least on laser cutters that I used, wide lanes would get engraved in raster mode, not in a vector mode. 

https://www.festi.info/boxes.py/BurnTest?FingerJoint_angle=90.0&FingerJoint_style=rectangular&FingerJoint_surroundingspaces=2.0&FingerJoint_edge_width=1.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&x=100&step=0.01&pairs=2&thickness=3.0&format=svg&tabs=0.0&debug=0&reference=100&burn=0.1&render=1 (`burn=0.1` - and all other parameters also at default)

[image]

https://www.festi.info/boxes.py/BurnTest?FingerJoint_angle=90.0&FingerJoint_style=rectangular&FingerJoint_surroundingspaces=2.0&FingerJoint_edge_width=1.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&x=100&step=0.01&pairs=2&thickness=3.0&format=svg&tabs=0.0&debug=0&reference=100&burn=0.4&render=1 (`burn=0.4` - and all other parameters at default)

[image]

florianfesti, 2021-03-01:
By default the lines have a width of two times the `burn` parameter. As a result the lines show the area that is expected to be cut away by the laser. So while the line is out-set the white area in the middle is the actual size.

This all assumes that the laser cutter is moving along the center of the lines. If your software needs thin lines you can use the svg_Ponoko format. If this also doesn't work please state the software you are using and the necessary formatting and I may just add another format.

matkoniecz, 2021-03-01:
> This all assumes that the laser cutter is moving along the center of the lines.

So lasers that I used so far were unusual! Thanks for an explanation!

closed 2021-03-01 by matkoniecz